cilynx/rtl88x2bu #133: Crash on kernel Linux gray-compute 5.10.0-14-generic

gadLinux, 2021-03-03:
The driver crashes when joining to the network on linux 5.10.

Linux gray-compute 5.10.0-14-generic #15-Ubuntu SMP Fri Jan 29 15:10:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

12:26:44 gray-compute NetworkManager[2194]: <info>  [1614770804.9033] device (wlx00e04c23275a): supplicant interface state: disconnected -> inactive
Mar  3 12:26:45 gray-compute /usr/libexec/gdm-x-session[5241]: (EE) client bug: timer event2 debounce short: scheduled expiry is in the past (-7ms), your system is too slow
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5403] device (wlx00e04c23275a): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5411] device (wlx00e04c23275a): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5415] device (wlx00e04c23275a): Activation: (wifi) connection 'AzulNetwork 2' has security, and secrets exist.  No new secrets needed.
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5416] Config: added 'ssid' value 'AzulNetwork'
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5416] Config: added 'scan_ssid' value '1'
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5416] Config: added 'bgscan' value 'simple:30:-70:86400'
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5416] Config: added 'key_mgmt' value 'SAE FT-SAE'
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5416] Config: added 'auth_alg' value 'OPEN'
Mar  3 12:26:48 gray-compute NetworkManager[2194]: <info>  [1614770808.5417] Config: added 'psk' value '<hidden>'
Mar  3 12:26:50 gray-compute NetworkManager[2194]: <info>  [1614770810.6726] device (wlx00e04c23275a): supplicant interface state: inactive -> scanning
Mar  3 12:26:56 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: Trying to associate with b8:be:f4:3a:96:a3 (SSID='AzulNetwork' freq=5180 MHz)
Mar  3 12:26:56 gray-compute wpa_supplicant[16880]: nl80211: kernel reports: Authentication algorithm number required
Mar  3 12:26:56 gray-compute NetworkManager[2194]: <info>  [1614770816.8626] device (wlx00e04c23275a): supplicant interface state: scanning -> associating
Mar  3 12:26:56 gray-compute kernel: [ 7753.371869] ------------[ cut here ]------------
Mar  3 12:26:56 gray-compute kernel: [ 7753.371902] WARNING: CPU: 6 PID: 55352 at net/wireless/nl80211.c:3388 nl80211_send_chandef+0x16a/0x180 [cfg80211]
Mar  3 12:26:56 gray-compute kernel: [ 7753.371903] Modules linked in: 88x2bu(OE) cfg80211 uhid rfcomm nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype br_netfilter xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_chain_nat nft_counter rbd libceph nf_tables nfnetlink bridge stp nfsv3 llc nfs_acl nfs lockd vboxnetadp(OE) grace nfs_ssc vboxnetflt(OE) fscache vboxdrv(OE) cmac algif_hash algif_skcipher af_alg overlay bnep binfmt_misc input_leds intel_rapl_msr intel_rapl_common sb_edac x86_pkg_temp_thermal intel_powerclamp kvm_intel kvm crct10dif_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper rapl snd_hda_codec_realtek snd_hda_codec_generic intel_cstate hp_wmi wmi_bmof sparse_keymap ledtrig_audio serio_raw snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_usb_audio btusb snd_hda_core btrtl btbcm snd_usbmidi_lib btintel soundwire_bus gspca_ov534 uvcvideo hid_sony snd_soc_core snd_hwdep
Mar  3 12:26:56 gray-compute kernel: [ 7753.371954]  joydev bluetooth gspca_main videobuf2_vmalloc videobuf2_memops snd_compress videobuf2_v4l2 videobuf2_common ac97_bus snd_pcm_dmaengine ecdh_generic snd_pcm hid_elecom razerkbd(OE) amdgpu ecc snd_seq_midi videodev snd_seq_midi_event mc snd_rawmidi iommu_v2 snd_seq gpu_sched ttm snd_seq_device snd_timer drm_kms_helper cec snd rc_core i2c_algo_bit fb_sys_fops syscopyarea sysfillrect sysimgblt soundcore ioatdma dca tpm_infineon mac_hid sch_fq_codel nf_nat_pptp nf_conntrack_pptp nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bonding coretemp msr parport_pc ppdev lp parport binder_linux ashmem_linux(CE) drm sunrpc ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear wacom hid_logitech ff_memless hid_generic usbhid hid ums_realtek uas usb_storage crc32_pclmul psmouse i2c_i801 firewire_ohci ahci i2c_smbus firewire_core mpt3sas lpc_ich libahci isci crc_itu_t e1000e libsas raid_class
Mar  3 12:26:56 gray-compute kernel: [ 7753.372010]  xhci_pci scsi_transport_sas xhci_pci_renesas wmi
Mar  3 12:26:56 gray-compute kernel: [ 7753.372014] CPU: 6 PID: 55352 Comm: RTW_CMD_THREAD Tainted: G        WC OE     5.10.0-14-generic #15-Ubuntu
Mar  3 12:26:56 gray-compute kernel: [ 7753.372015] Hardware name: Hewlett-Packard HP Z820 Workstation/158B, BIOS J63 v03.95 06/13/2019
Mar  3 12:26:56 gray-compute kernel: [ 7753.372030] RIP: 0010:nl80211_send_chandef+0x16a/0x180 [cfg80211]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372036] Code: 4d dc ba 04 00 00 00 4c 89 e7 89 45 dc be a1 00 00 00 e8 b9 fb 6f e6 85 c0 74 82 41 bd 97 ff ff ff e9 77 ff ff ff 31 c0 eb af <0f> 0b 41 bd ea ff ff ff e9 66 ff ff ff e8 14 cc d2 e6 0f 1f 40 00
Mar  3 12:26:56 gray-compute kernel: [ 7753.372037] RSP: 0018:ffffb00fc27c3d18 EFLAGS: 00010246
Mar  3 12:26:56 gray-compute kernel: [ 7753.372038] RAX: 0000000000000000 RBX: ffffb00fc27c3de8 RCX: 0000000000000000
Mar  3 12:26:56 gray-compute kernel: [ 7753.372039] RDX: 0000000000000004 RSI: ffffb00fc27c3de8 RDI: ffff99300e64a868
Mar  3 12:26:56 gray-compute kernel: [ 7753.372040] RBP: ffffb00fc27c3d40 R08: 0000000000000eac R09: ffff992f71f5501c
Mar  3 12:26:56 gray-compute kernel: [ 7753.372040] R10: 0000000000000000 R11: 0000000000000021 R12: ffff99301e4ea000
Mar  3 12:26:56 gray-compute kernel: [ 7753.372041] R13: ffffb00fc27c3de8 R14: ffff99301e4ea000 R15: ffff992f71f55014
Mar  3 12:26:56 gray-compute kernel: [ 7753.372042] FS:  0000000000000000(0000) GS:ffff994b6fd80000(0000) knlGS:0000000000000000
Mar  3 12:26:56 gray-compute kernel: [ 7753.372043] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar  3 12:26:56 gray-compute kernel: [ 7753.372044] CR2: 00003b84fd233ff0 CR3: 0000002003c10001 CR4: 00000000000606e0
Mar  3 12:26:56 gray-compute kernel: [ 7753.372045] Call Trace:
Mar  3 12:26:56 gray-compute kernel: [ 7753.372051]  ? __nla_put+0x20/0x30
Mar  3 12:26:56 gray-compute kernel: [ 7753.372066]  nl80211_ch_switch_notify.constprop.0+0xce/0x170 [cfg80211]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372080]  cfg80211_ch_switch_started_notify+0x4a/0x90 [cfg80211]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372124]  rtw_cfg80211_ch_switch_notify+0x112/0x140 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372147]  rtw_chk_start_clnt_join+0x66/0x80 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372181]  join_cmd_hdl+0x220/0x380 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372202]  rtw_cmd_thread+0x374/0x450 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372229]  ? rtw_chk_start_clnt_join+0x80/0x80 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372233]  kthread+0x12f/0x150
Mar  3 12:26:56 gray-compute kernel: [ 7753.372255]  ? rtw_stop_cmd_thread+0x50/0x50 [88x2bu]
Mar  3 12:26:56 gray-compute kernel: [ 7753.372257]  ? __kthread_bind_mask+0x70/0x70
Mar  3 12:26:56 gray-compute kernel: [ 7753.372261]  ret_from_fork+0x22/0x30
Mar  3 12:26:56 gray-compute kernel: [ 7753.372264] ---[ end trace 2d499ce711e93e0c ]---
Mar  3 12:26:57 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-ASSOC-REJECT status_code=40
Mar  3 12:26:57 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="AzulNetwork" auth_failures=1 duration=10 reason=CONN_FAILED
Mar  3 12:26:57 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Mar  3 12:26:57 gray-compute NetworkManager[2194]: <info>  [1614770817.0183] device (wlx00e04c23275a): supplicant interface state: associating -> disconnected
Mar  3 12:26:58 gray-compute systemd-resolved[2023]: Server returned error NXDOMAIN, mitigating potential DNS violation DVE-2018-0001, retrying transaction with reduced feature level UDP.
[... cut ...]
Mar  3 12:27:40 gray-compute kernel: [ 7797.171398]  ret_from_fork+0x22/0x30
Mar  3 12:27:40 gray-compute kernel: [ 7797.171400] ---[ end trace 2d499ce711e93e0e ]---
Mar  3 12:27:40 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-ASSOC-REJECT status_code=40
Mar  3 12:27:40 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="AzulNetwork" auth_failures=2 duration=20 reason=CONN_FAILED
Mar  3 12:27:40 gray-compute wpa_supplicant[16880]: wlx00e04c23275a: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

MaxG87, 2021-03-08:
Sorry, I cannot reproduce that. I use the driver on both Ubuntu and Debian without any issues.

Does the driver work with different kernels or different hardware?

vito-r1, 2021-03-16:
Hello,
I have a similar issue. Please check my log:
Currently it happens when I connect it trough usb hub

Mar 16 12:30:51  systemd-networkd[1679]: wlxd03745c04952: Link UP
Mar 16 12:30:51  systemd-udevd[23911]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Mar 16 12:30:51  wpa_supplicant[1777]: nl80211: kernel reports: Authentication algorithm number required
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.5949] device (wlxd03745c04952): supplicant interface state: internal-starting -> disconnected
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.5955] Wi-Fi P2P device controlled by interface wlxd03745c04952 created
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.5963] manager: (p2p-dev-wlxd03745c04952): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/30)
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.6017] device (p2p-dev-wlxd03745c04952): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Mar 16 12:30:51  NetworkManager[1756]: <warn>  [1615894251.6070] sup-iface[22b13517e64f97bf,13,wlxd03745c04952]: call-p2p-cancel: failed with P2P cancel failed
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.6074] device (wlxd03745c04952): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Mar 16 12:30:51  NetworkManager[1756]: <info>  [1615894251.6131] device (p2p-dev-wlxd03745c04952): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.2871] policy: auto-activating connection 'p' (ab4)
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.2878] device (wlxd03745c04952): Activation: starting connection 'p' (ab4)
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.2880] device (wlxd03745c04952): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.2890] manager: NetworkManager state is now CONNECTING
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.2971] device (wlxd03745c04952): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3039] device (wlxd03745c04952): Activation: (wifi) access point 'p' has security, but secrets are required.
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3040] device (wlxd03745c04952): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3154] device (wlxd03745c04952): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3159] device (wlxd03745c04952): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3162] device (wlxd03745c04952): Activation: (wifi) connection 'p' has security, and secrets exist.  No new secrets needed.
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'ssid' value 'p'
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'scan_ssid' value '1'
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'bgscan' value 'simple:30:-70:86400'
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'auth_alg' value 'OPEN'
Mar 16 12:30:55  NetworkManager[1756]: <info>  [1615894255.3163] Config: added 'psk' value '<hidden>'
Mar 16 12:30:55  wpa_supplicant[1777]: wlxd03745c04952: Trying to associate with b4 (SSID='p' freq=2462 MHz)
Mar 16 12:30:55  wpa_supplicant[1777]: nl80211: kernel reports: Authentication algorithm number required
Mar 16 12:30:55  systemd[1]: systemd-rfkill.service: Succeeded.
Mar 16 12:30:56  kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Mar 16 12:30:56  kernel: ------------[ cut here ]------------
Mar 16 12:30:56  kernel: WARNING: CPU: 0 PID: 23917 at net/wireless/nl80211.c:3388 nl80211_send_chandef+0x175/0x190 [cfg80211]
Mar 16 12:30:56  kernel: Modules linked in: 88x2bu(OE) cdc_ether usbnet r8152 mii nls_utf8 isofs cdrom uas usb_storage rfcomm bridge ctr ccm rtsx_usb_ms memstick cmac algif_hash algif_skcipher af_alg bnep btusb btrtl btbcm btintel bluetooth jitterentropy_rng uvcvideo videobuf2_vmalloc videobuf2_memops rtl8xxxu videobuf2_v4l2 videobuf2_common drbg videodev mac80211 ansi_cprng ecdh_generic libarc4 ecc mc lz4 zram zsmalloc nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter xt_CHECKSUM xt_tcpudp nft_compat nf_tables nfnetlink cfg80211 rfkill cpufreq_ondemand nls_ascii snd_hda_codec_hdmi nls_cp437 vfat fat snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_sof_xtensa_dsp snd_sof snd_sof_intel_hda snd_soc_skl x86_pkg_temp_thermal intel_powerclamp snd_soc_hdac_hda kvm_intel snd_hda_ext_core snd_soc_sst_ipc snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi intel_rapl_msr snd_hda_intel
[... cut ...]
Mar 16 12:30:56  kernel:  rtw_chk_start_clnt_join+0x62/0x70 [88x2bu]
Mar 16 12:30:56  kernel:  join_cmd_hdl+0x1fb/0x360 [88x2bu]
Mar 16 12:30:56  kernel:  rtw_cmd_thread+0x370/0x450 [88x2bu]
Mar 16 12:30:56  kernel:  ? rtw_chk_start_clnt_join+0x70/0x70 [88x2bu]
Mar 16 12:30:56  kernel:  ? rtw_stop_cmd_thread+0x40/0x40 [88x2bu]
Mar 16 12:30:56  kernel:  kthread+0x11b/0x140
Mar 16 12:30:56  kernel:  ? __kthread_bind_mask+0x60/0x60
Mar 16 12:30:56  kernel:  ret_from_fork+0x1f/0x30
Mar 16 12:30:56  kernel: ---[ end trace ce0ed56c4d136dea ]---

tedliosu, 2021-06-02:
I believe I have a similar issue as well, especially as I've noticed that all of these kernel warnings contain call traces similar to the following I keep getting on my machine in my dmesg output:
`[   21.012223] Call Trace:
[   21.012230]  ? __nla_put+0x20/0x30
[   21.012260]  nl80211_ch_switch_notify.constprop.0+0xce/0x170 [cfg80211]
[   21.012287]  cfg80211_ch_switch_started_notify+0x51/0xa0 [cfg80211]
[   21.012361]  rtw_cfg80211_ch_switch_notify+0x114/0x140 [88x2bu]
[   21.012413]  rtw_chk_start_clnt_join+0x66/0x80 [88x2bu]
[   21.012463]  join_cmd_hdl+0x220/0x380 [88x2bu]
[   21.012502]  rtw_cmd_thread+0x30e/0x3f0 [88x2bu]
[   21.012553]  ? rtw_chk_start_clnt_join+0x80/0x80 [88x2bu]
[   21.012557]  kthread+0x114/0x150
[   21.012595]  ? rtw_stop_cmd_thread+0x40/0x40 [88x2bu]
[   21.012597]  ? kthread_park+0x90/0x90
[   21.012602]  ret_from_fork+0x22/0x30`

And this was bugging me to no end as I've tried updating to the latest driver (_5.8.7.1_35809.20191129_COEX20191120-7777_) but now every time I start-up my computer that darn call trace would re-appear in my dmesg output.

So I dug around online for a patch that could resolve these kernel warnings and not have them appear again, and I happened to stumble upon a potential fix for this problem [here](https://github.com/aircrack-ng/rtl8812au/issues/660#issuecomment-645636370).  Now I understand that the issue linked is for a different wifi chipset, but when I tried applying **the same fix** to the os_dep/linux/ioctl_cfg80211.c file at around line 430 and re-deployed the patched driver, voila the kernel warnings disappeared from my journalctl and dmesg outputs.  The patch works for both kernel 5.4.0-73-generic and kernel 5.8.0-53-generic as I've tested on my machine.

So please try the fix I've linked here and y'all let me know if it works for you too.  I'll also be more than happy to fork this repo and apply the fix so that the problem may be resolved via a pull request if this solution works for everyone else.

MaxG87, 2021-06-02:
Thank you for the hint!

I created Branch fix-133 (https://github.com/cilynx/rtl88x2bu/tree/fix-133) where I initialised the struct in question. Could you please check if it works and resolves the call trace issue for you?

tedliosu, 2021-06-02:
> Thank you for the hint!
> 
> I created Branch fix-133 (https://github.com/cilynx/rtl88x2bu/tree/fix-133) where I initialised the struct in question. Could you please check if it works and resolves the call trace issue for you?

## For Ubuntu Distro Running Kernel 5.4.0-73-generic ##

Dmesg output **before** _fix-133_ is applied:

> [   13.877550] Bluetooth: RFCOMM socket layer initialized
> [   13.877557] Bluetooth: RFCOMM ver 1.11
> [   24.372352] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
> [   24.628940] ------------[ cut here ]------------
> [   24.628969] WARNING: CPU: 0 PID: 1041 at net/wireless/nl80211.c:3160 nl80211_send_chandef+0x142/0x160 [cfg80211]
> [   24.628970] Modules linked in: rfcomm cmac algif_hash algif_skcipher af_alg aufs bnep overlay btusb btrtl btbcm btintel sch_fq_codel bluetooth ecdh_generic ecc msr binfmt_misc nls_iso8859_1 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep edac_mce_amd snd_pcm kvm_amd snd_seq_midi snd_seq_midi_event kvm snd_rawmidi snd_seq crct10dif_pclmul ghash_clmulni_intel snd_seq_device snd_timer aesni_intel crypto_simd cryptd wmi_bmof joydev glue_helper input_leds snd k10temp ccp soundcore mac_hid 88x2bu(OE) cfg80211 nct6775 hwmon_vid parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid amdgpu(OE) amd_iommu_v2 amd_sched(OE) amdttm(OE) amdkcl(OE) drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops crc32_pclmul drm i2c_piix4 nvme r8169 i2c_algo_bit ahci realtek nvme_core libahci wmi gpio_amdpt gpio_generic
> [   24.629018] CPU: 0 PID: 1041 Comm: RTW_CMD_THREAD Tainted: G           OE     5.4.0-73-generic #82-Ubuntu
> [   24.629020] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./B450M Pro4, BIOS P3.60 07/31/2019
> [   24.629043] RIP: 0010:nl80211_send_chandef+0x142/0x160 [cfg80211]
> [   24.629046] Code: 4d dc ba 04 00 00 00 4c 89 e7 89 45 dc be a1 00 00 00 e8 11 cc 83 df 85 c0 74 82 41 bd 97 ff ff ff e9 77 ff ff ff 31 c0 eb af <0f> 0b 41 bd ea ff ff ff e9 66 ff ff ff e8 dc b5 39 df 66 66 2e 0f
> [   24.629048] RSP: 0018:ffffa6a1814a3d18 EFLAGS: 00010246
> [   24.629050] RAX: 0000000000000000 RBX: ffffa6a1814a3df0 RCX: 00000000000000a1
> [   24.629051] RDX: 0000000080531834 RSI: 00000000ffff3080 RDI: ffffa6a1814a3df0
> [   24.629052] RBP: ffffa6a1814a3d40 R08: 0000000000000000 R09: ffff927406cdf01c
> [   24.629053] R10: 0000000000000000 R11: 000000000000001f R12: ffff927402fe6e00
> [   24.629054] R13: ffffa6a1814a3df0 R14: ffff927402fe6e00 R15: ffff927406cdf014
> [   24.629056] FS:  0000000000000000(0000) GS:ffff92740e800000(0000) knlGS:0000000000000000
> [   24.629057] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
> [   24.629058] CR2: 0000558fc47052d8 CR3: 0000000405844000 CR4: 00000000003406f0
> [   24.629060] Call Trace:
> [   24.629067]  ? __nla_put+0x20/0x30
> [   24.629088]  nl80211_ch_switch_notify.constprop.0+0xcf/0x170 [cfg80211]
> [   24.629108]  cfg80211_ch_switch_started_notify+0x51/0xa0 [cfg80211]
> [   24.629172]  rtw_cfg80211_ch_switch_notify+0x110/0x140 [88x2bu]
> [   24.629214]  rtw_chk_start_clnt_join+0x66/0x80 [88x2bu]
> [   24.629256]  join_cmd_hdl+0x220/0x380 [88x2bu]
> [   24.629286]  rtw_cmd_thread+0x30e/0x3f0 [88x2bu]
> [   24.629328]  ? rtw_chk_start_clnt_join+0x80/0x80 [88x2bu]
> [   24.629332]  kthread+0x104/0x140
> [   24.629362]  ? rtw_stop_cmd_thread+0x40/0x40 [88x2bu]
> [   24.629364]  ? kthread_park+0x90/0x90
> [   24.629367]  ret_from_fork+0x22/0x40
> [   24.629370] ---[ end trace bb9870ec421b4a1c ]---
> [   25.048408] IPv6: ADDRCONF(NETDEV_CHANGE): wlx34e8945ff796: link becomes ready
> [   25.485305] audit: type=1400 audit(1622671708.519:32): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=1249 comm="apparmor_parser"
> [   25.552019] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
> [   25.554197] Bridge firewalling registered
> [   25.562695] bpfilter: Loaded bpfilter_umh pid 1254
> [   25.562903] Started bpfilter
> [   25.634061] Initializing XFRM netlink socket

Dmesg output **after** _fix-133_ is applied:

> [    4.328563] EDAC amd64: Node 0: DRAM ECC disabled.
> [    4.328564] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
>                 Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
>                 (Note that use of the override may cause unknown side effects.)
> [    7.955602] usbcore: registered new interface driver rtl88x2bu
> [    7.970155] rtl88x2bu 2-2:1.0 wlx34e8945ff796: renamed from wlan0
> [    7.982582] Bluetooth: Core ver 2.22
> [    7.982606] NET: Registered protocol family 31
> [    7.982607] Bluetooth: HCI device and connection manager initialized
> [    7.982612] Bluetooth: HCI socket layer initialized
> [    7.982613] Bluetooth: L2CAP socket layer initialized
> [    7.982615] Bluetooth: SCO socket layer initialized
> [    7.989151] usbcore: registered new interface driver btusb
> [    8.105577] Bluetooth: hci0: BCM: chip id 63
> [    8.108577] Bluetooth: hci0: BCM: features 0x07
> [    8.126584] Bluetooth: hci0: opencl-os
> [    8.128586] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1764
> [    8.133297] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[... cut ...]
> [   47.588471] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
> [   80.588756] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
> [  123.620494] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
> [  156.975516] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
> [  165.998408] IPv6: ADDRCONF(NETDEV_CHANGE): wlx34e8945ff796: link becomes ready

MaxG87, 2021-06-03:
The output of dmesg looks much improved. Does the driver still work - can you surf the web with it? I cannot test that in the moment.

MaxG87, 2021-06-03:
Fixed by 3a08c6f50107f75ec2b894e00db0c57c8fab6b5a. I applied the patch anyways.

closed 2021-06-03 by MaxG87