cilynx/rtl88x2bu #134: Issue building module for 5.10.0-kali3-amd64

javilinux, 2021-03-04:
Following the instructions in the build step I get the following error:

```
sudo dkms build -m rtl88x2bu -v ${VER}       

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.0-kali3-amd64 KVER=5.10.0-kali3-amd64 src=/usr/src/rtl88x2bu-5.6.1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.

```
The last lines of /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log are :

```
/usr/src/linux-headers-5.10.0-kali3-common/include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.build:284: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.build:284: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o] Error 1
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.build:284: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o] Error 1
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.build:284: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:1817: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-kali3-amd64'
make: *** [Makefile:2288: modules] Error 2
```

Lsusb information:
Bus 002 Device 007: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

MaxG87, 2021-03-08:
Which revision are you compiling? The `sha256_state` bug should be fixed on the 5.6.1.6 and 5.8.1 branches.

UbuntuStrike, 2021-03-15:
5.10.0-kali4-amd64 is working with version 5.6.1... just make sure to "apt install bc" prior

vito-r1, 2021-03-17:
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-4-amd64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [/usr/src/linux-headers-5.10.0-4-common/Makefile:717: include/config/auto.conf] Error 1
make[1]: *** [/usr/src/linux-headers-5.10.0-4-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-4-amd64'
make: *** [Makefile:2370: modules] Error 2

Linux version 5.10.0-4-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Debian 5.10.19-1 (2021-03-02)

Hello ,
I have received this error on both versions 5.8.7 and 5.6.1. Could you check it and help me to build this module.
Thank you

closed 2021-03-29 by MaxG87