cilynx/rtl88x2bu #138: Version 5.6.1.6_x working, version 5.8.7.1_x not

JakubOrsula, 2021-03-14:
Hi,
first, many thanks for providing the driver. Here's some more info.
I'm running freshly installed linux mint xfce with kernel: 5.8.0-44-generic #50~20.04.1-Ubuntu
My device is tp-link T3U Plus. It works nicely with 5.6.1.6 version but can't connect to wifi on  5.8.7.1.

I would like to help you out with logs, but I don't know which one would interest you.

MaxG87, 2021-03-29:
Thank you for reaching out.
Does the driver compile at least? If not, I would be interested in the build log. If it compiles, could you look if the last lines `sudo dmesg -T` look related?
However, I have to admit that I only resolve compile issues. I do not understand the inner workings and usually can't fix them.

JakubOrsula, 2021-03-29:
The driver compiled all and well. I used another driver which worked, and now I fear that reinstalling this one might bring the issues back. My apologies but we can probably close this.

closed 2021-03-29 by JakubOrsula