cilynx/rtl88x2bu #142: dkms does not work on branch 5.8.7.1 due to KERNELRELEASE variable.

Houndie, 2021-03-21:
Using ./deploy.sh does not work on latest debian release due to dkms silently adding the KERNELRELEASE flag:

```
make -j2 KERNELRELEASE=4.19.0-14-amd64 KVER=4.19.0-14-amd64 src=/sr/src/rtl88x2bu-5.8.7.1
make: *** No targets. Stop
```

From the dkms manfile:

```
KERNELRELEASE  will  be automatically appended to MAKE[#].  If you want to suppress
              this behavior, you can quote the make command: 'make'.
```

Suggest quoting make in dkms.conf?  It seems to work in this case.

MaxG87, 2021-03-29:
Thank you for your report.
It seems for me that once one ran `./deploy.sh` successfully, DKMS can update this driver. At least, I did not noticed any glitches on Bullseye.
I will have to inspect the issue locally. I will comment here again once done.

MaxG87, 2021-04-24:
You are right, the KERNELRELEASE variable breaks things here.

The fault is more on the Makefiles side, because it includes the targets only if KERNELRELEASE is unset. Very weird.

I am about to create a patch for this, which will do exactly what you proposed: quoting `make`. Thank you for your contribution.

closed 2021-04-24 by MaxG87