cilynx/rtl88x2bu #144: [5.8.7.1] It there RPI support?

mdziezyc, 2021-04-05:
README.md in 5.8.7.1 in section [Raspberry Pi Access Point](https://github.com/cilynx/rtl88x2bu/tree/5.8.7.1_35809.20191129_COEX20191120-7777#raspberry-pi-access-point) states `sed -i 's/ARM_RPI = n/ARM_RPI = y/' Makefile`, but in the makefile there is no `ARM_RPI` which is in [default branch](https://github.com/cilynx/rtl88x2bu/blob/48e7c19c92a77554403e1347447f8e2cfd780228/Makefile#L117) (5.6.1).

This probably leads to error on my RPI4B:
```
pi@raspberrypi:~/rtl88x2bu $ sudo dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.17-v7l+ KVER=5.10.17-v7l+ src=/usr/src/rtl88x2bu-5.8.7.1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.17-v7l+ (armv7l)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log for more information.
pi@raspberrypi:~/rtl88x2bu $ cat /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log
DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 5.10.17-v7l+ (armv7l)
Mon 05 Apr 2021 08:17:30 PM CEST
make: *** No targets.  Stop.
```

MaxG87, 2021-04-13:
Thank you very much for pointing out. This needs to be fixed.

Can you compile the driver for [5.6.1.6](https://github.com/cilynx/rtl88x2bu/tree/5.6.1.6_35492.20191025_COEX20180928-6a6a) on a Raspberry Pi?

I will set up my RPI for this and investigate this problem.

MaxG87, 2021-04-15:
I created PR #148 which should address the issue. I would appreciate to get some feedback here.

mdziezyc, 2021-04-16:
Sorry, but I bricked my RPi somehow, so I cannot check it right now. I'll get back to you if I managed to get it going again.

MaxG87, 2021-04-16:
What RPi do you use? I used a RPi 4 with 32bit Raspbian. If your setup is usual I might just merge the PR.

MaxG87, 2021-04-24:
I just merged the PR. Please open a new issue if your problem is not resolved by it.

closed 2021-04-24 by MaxG87