cilynx/rtl88x2bu #151: kernel 5.12

joksik, 2021-05-06:
Hi, i use arch linux and today we had an kernel update.
with new kernel i have error in dkms install.
my logs:

` CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_cfg80211.o
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                       ^~~~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c:358:55: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_cfgvendor.o
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o] Błąd 1
make[2]: *** Oczekiwanie na niezakończone zadania....
make[1]: *** [Makefile:1851: /var/lib/dkms/rtl88x2bu/5.6.1/build] Błąd 2
make[1]: Opuszczenie katalogu '/usr/lib/modules/5.12.1-arch1-1/build'
make: *** [Makefile:2284: modules] Błąd 2
`

CGarces, 2021-05-06:
Was fixed on 8316868608bd0fc3de48dd7d5d7ebe4648f5a181
Please use 5.8.7.1_35809.20191129_COEX20191120-7777 branch (or send a PR and apply the patch)

joksik, 2021-05-17:
hi, it was work for 2 weeks but after today update to kernel 5.12.4 even this branch 5.8.7.1_35809.20191129_COEX20191120-7777 stop working. in logs i have something like this:

``DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 5.12.4-arch1-2 (x86_64)
pon, 17 maj 2021, 14:09:04 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.12.4-arch1-2/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Wejście do katalogu '/usr/lib/modules/5.12.4-arch1-2/build'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme.o
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme.c:3454:17: error: this ‘else’ clause does not guard... [-Werror=misleading-indentation]
 3454 |                 else
      |                 ^~~~
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme.c:3457:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3457 |                         goto exit;
      |                         ^~~~
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme.o] Błąd 1
make[1]: *** [Makefile:1851: /var/lib/dkms/rtl88x2bu/5.8.7.1/build] Błąd 2
make[1]: Opuszczenie katalogu '/usr/lib/modules/5.12.4-arch1-2/build'
make: *** [Makefile:2388: modules] Błąd 2

CGarces, 2021-05-17:
Hi

I have send # to try to reproduce your error without luck.

See my last build using 5.12.4
https://github.com/CGarces/rtl88x2bu/runs/2601653133?check_suite_focus=true

It compiles without errors.

hood, 2021-05-17:
I'm here to report the build error `error: this ‘else’ clause does not guard...` in `rtwl_mlme.c` is happening on branch `5.6.1.6_35492.20191025_COEX20180928-6a6a` too.
System info: `Linux arch 5.12.4-arch1-2 #1 SMP PREEMPT Sat, 15 May 2021 20:58:02 +0000 x86_64 GNU/Linux`.

pimpeters, 2021-05-20:
> I'm here to report the build error `error: this ‘else’ clause does not guard...` in `rtwl_mlme.c` is happening on branch `5.6.1.6_35492.20191025_COEX20180928-6a6a` too.
> System info: `Linux arch 5.12.4-arch1-2 #1 SMP PREEMPT Sat, 15 May 2021 20:58:02 +0000 x86_64 GNU/Linux`.
> 
> EDIT
> Found out the issue. GCC11 is being shipped in Core for Arch Linux users. Updating to GCC11 makes builds fail for the reason listed before.

Did you find any way to fix this? I'm getting the same error now.

hood, 2021-05-20:
You can apply this temporary patch of mine to fix the issue. I'm not submitting it as a PR because I've seen something similar is already in waiting of approval: https://github.com/cilynx/rtl88x2bu/pull/150.

```
diff --git a/core/efuse/rtw_efuse.c b/core/efuse/rtw_efuse.c
index 59b3464..d5c1909 100644
--- a/core/efuse/rtw_efuse.c
+++ b/core/efuse/rtw_efuse.c
@@ -12,6 +12,8 @@
  * more details.
  *
  *****************************************************************************/
+#pragma GCC diagnostic ignored "-Wmisleading-indentation"
+
 #define _RTW_EFUSE_C_
 
 #include <drv_types.h>
diff --git a/core/rtw_mlme.c b/core/rtw_mlme.c
index cc83c14..73292ad 100644
--- a/core/rtw_mlme.c
+++ b/core/rtw_mlme.c
@@ -12,6 +12,8 @@
  * more details.
  *
  *****************************************************************************/
+#pragma GCC diagnostic ignored "-Wmisleading-indentation"
+
 #define _RTW_MLME_C_
 
 #include <hal_data.h>
diff --git a/core/rtw_wlan_util.c b/core/rtw_wlan_util.c
index f2c3228..9432ea4 100644
--- a/core/rtw_wlan_util.c
+++ b/core/rtw_wlan_util.c
@@ -12,6 +12,8 @@
  * more details.
  *
  *****************************************************************************/
+#pragma GCC diagnostic ignored "-Wmisleading-indentation"
+
 #define _RTW_WLAN_UTIL_C_
 
 #include <drv_types.h>
```

farribeiro, 2021-05-23:
I having trouble with the lastest commit of 5.8.7.1 with new fedora stable kernel 5.12  on my Silverblue... i created a copr repo using kmod: https://copr.fedorainfracloud.org/coprs/farribeiro/rtl88x2bu-kmod/

the log trying to install the module

```mai 23 10:13:04 i7 rpm-ostree(akmod-rtl88x2bu.post)[58018]: Building /usr/src/akmods/rtl88x2bu-kmod-5.8.7.1-1.fc34.src.rpm for kernel 5.12.5-300.fc34.x86_64
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_ieee80211.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme_ext.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c: In function 'rtw_drv_scan_by_self':
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c:3454:17: error: this 'else' clause does not guard... [-Werror=misleadin>
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:  3454 |                 else
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:       |                 ^~~~
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c:3457:25: note: ...this statement, but the latter is misleadingly indent>
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:  3457 |                         goto exit;
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:       |                         ^~~~
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mi.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_wlan_util.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_vht.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_pwrctrl.o
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: cc1: all warnings being treated as errors
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[2]: *** [scripts/Makefile.build:271: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.o] Error 1
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[2]: *** Waiting for unfinished jobs....
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[1]: *** [Makefile:1852: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64] Error 2
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[1]: Leaving directory '/usr/src/kernels/5.12.5-300.fc34.x86_64'
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make: *** [Makefile:2388: modules] Error 2
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: erro: Status de saída de /var/tmp/rpm-tmp.LrOAeO inválido (%build)
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: Erros na construção do RPM:
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Unable to open sqlite database /usr/share/rpm/rpmdb.sqlite: unable to open database file
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     cannot open Packages index using sqlite - Operação não permitida (1)
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     não foi possível abrir o banco de dados de pacotes em /usr/share/rpm
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     user mockbuild does not exist - using root
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     group mock does not exist - using root
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     user mockbuild does not exist - using root
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     group mock does not exist - using root
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Unable to open sqlite database /usr/share/rpm/rpmdb.sqlite: unable to open database file
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     cannot open Packages index using sqlite - Operação não permitida (1)
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     não foi possível abrir o banco de dados de pacotes em /usr/share/rpm
mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Status de saída de /var/tmp/rpm-tmp.LrOAeO inválido (%build)
```

farribeiro, 2021-05-24:
> I having trouble with the lastest commit of 5.8.7.1 with new fedora stable kernel 5.12 on my Silverblue... i created a copr repo using kmod: https://copr.fedorainfracloud.org/coprs/farribeiro/rtl88x2bu-kmod/ and my spec file in pagure.io https://pagure.io/rtl88x2bu-kmod
> 
> the log trying to install the module
> 
> ```
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_ieee80211.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme_ext.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c: In function 'rtw_drv_scan_by_self':
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c:3454:17: error: this 'else' clause does not guard... [-Werror=misleadin>
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:  3454 |                 else
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:       |                 ^~~~
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.c:3457:25: note: ...this statement, but the latter is misleadingly indent>
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:  3457 |                         goto exit;
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:       |                         ^~~~
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mi.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_wlan_util.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_vht.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:   CC [M]  /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_pwrctrl.o
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: cc1: all warnings being treated as errors
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[2]: *** [scripts/Makefile.build:271: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64/core/rtw_mlme.o] Error 1
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[2]: *** Waiting for unfinished jobs....
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[1]: *** [Makefile:1852: /tmp/akmodsbuild.1fCge64L/BUILD/rtl88x2bu-kmod-5.8.7.1/_kmod_build_5.12.5-300.fc34.x86_64] Error 2
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make[1]: Leaving directory '/usr/src/kernels/5.12.5-300.fc34.x86_64'
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: make: *** [Makefile:2388: modules] Error 2
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: erro: Status de saída de /var/tmp/rpm-tmp.LrOAeO inválido (%build)
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]: Erros na construção do RPM:
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Unable to open sqlite database /usr/share/rpm/rpmdb.sqlite: unable to open database file
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     cannot open Packages index using sqlite - Operação não permitida (1)
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     não foi possível abrir o banco de dados de pacotes em /usr/share/rpm
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     user mockbuild does not exist - using root
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     group mock does not exist - using root
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     user mockbuild does not exist - using root
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     group mock does not exist - using root
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Unable to open sqlite database /usr/share/rpm/rpmdb.sqlite: unable to open database file
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     cannot open Packages index using sqlite - Operação não permitida (1)
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     não foi possível abrir o banco de dados de pacotes em /usr/share/rpm
> mai 23 10:13:13 i7 rpm-ostree(akmod-rtl88x2bu.post)[58605]:     Status de saída de /var/tmp/rpm-tmp.LrOAeO inválido (%build)
> ```

I fixed that error pointing in the makefile that patch...

https://pagure.io/rtl88x2bu-kmod/blob/master/f/flags.patch

MaxG87, 2021-05-31:
Fixed by ded19c802d5817887ad7fea6fc99e720f4074f6e. Will be addressed further by #150 or #153.

closed 2021-05-31 by MaxG87