cilynx/rtl88x2bu #154: Should we remove support for DWA 183?

MaxG87, 2021-06-12:
My DWA-182 broke and I need a new 5Ghz dongle. Luckily for this project it seems as if there are no devices with internal antenna that have their driver in the kernel tree, so I will pick one that works with rtl88x2bu and continue to fix compilation bugs :expressionless: . 

However, I was unable to find the DLINK DWA-183. All sites that mention it are somewhat dubious. Assuming I did not miss anything important, there are two options:

1. The device does not exist anymore.
2. The comment behind the device ID is incorrect.

In both cases I would like to delete this line to avoid confusions and conflicts with other devices.

@cilynx What do you think. Should we delete that line?

cilynx, 2021-09-04:
I think DWA-183 is legit, but only available in Taiwan.  If you go to https://www.dlink.com/en/change-region and select Taiwan, it takes you to http://www.dlinktw.com.tw/.  Why they don't just use dlink.com.tw, who knows -- looks like they do something similar with Latin America...gotta love consistency.  Anyways, if you search for DWA-183 on the Taiwan regional site, the top result is http://www.dlinktw.com.tw/home/product?id=5087 which appears to be a legit product page.

MaxG87, 2021-09-06:
That sounds very reasonable. Then we can keep this definition.

closed 2021-09-06 by MaxG87