cilynx/rtl88x2bu #156: Install on Opensuse Leap15.3

hornetster, 2021-06-16:
Trying to install on Opensuse Leap 15.3, kernel 5.3.18-57-default and getting:
```
make -j4 KERNELRELEASE=5.3.18-57-default KVER=5.3.18-57-default src=/usr/src/rtl88x2bu-5.6.1.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.3.18-57-default (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
```
[make.log](https://github.com/cilynx/rtl88x2bu/files/6662263/make.log)

arnoldix, 2021-06-27:
This workaround helped on my system:
Modify the file os_dep/linux/ioctl_cfg80211.c. You will find several times this line
#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0))
or
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))

Search and replace KERNEL_VERSION(5, 8, 0) by KERNEL_VERSION(5, 3, 0) and re-run make.

Remark: In Leap 15.2 the kernel version is 5.3.18 as well and the make command runs without the described modification. Obviously SUSE applied a change in Leap 15.3 in the kernel 5.3.18 that other did in the kernel 5.8.0.

hornetster, 2021-06-28:
Thanks for the suggestion.
I ended up finding some software on the Opensuse (https://software.opensuse.org/search?utf8=%E2%9C%93&baseproject=ALL&q=rtl88) and that worked fine!

closed 2021-06-28 by hornetster