cilynx/rtl88x2bu #157: GRO_DROP is not compatible with kernel 5.12

MatthewForrester, 2021-06-24:
Building the 5.8.7.1_35809.20191129_COEX20191120-7777 branch failed on Fedora 34 with kernel 5.12.

Here are the most relevant parts of the build log:

```make
home/matthew/Repos/rtl88x2bu/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/home/matthew/Repos/rtl88x2bu/os_dep/linux/recv_linux.c:358:76: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |                         if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                                            ^~~~~~~~
/home/matthew/Repos/rtl88x2bu/os_dep/linux/recv_linux.c:358:76: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:271: /home/matthew/Repos/rtl88x2bu/os_dep/linux/recv_linux.o] Error 1
make[1]: *** [Makefile:1857: /home/matthew/Repos/rtl88x2bu] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.12.11-300.fc34.x86_64'
make: *** [Makefile:2284: modules] Error 2
```

I think that this bug may be fixed by [this commit](https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/657b7cfde9958e273febdeaeac579902e407f577) in a sister repository.

Debugging drivers is way out of my league, but hopefully this is helpful feedback.

MaxG87, 2021-07-06:
Thank you for your report. I am currently building v5.13. I will have a look at the problem once that finished.

MaxG87, 2021-07-06:
I cannot reproduce and cannot understand how that happens to you. The problem described by you is fixed in 8316868608bd0fc3de48dd7d5d7ebe4648f5a181 which was commited in March.

Please ensure to use the latest revision, that is 1f859429e8bc42d888fd88c1aecd512c4386f2b2.

closed 2021-07-06 by MaxG87