cilynx/rtl88x2bu #158: Is it possible to avoid rebuilding of the module via DKMS at every kernel update?

All3xJ, 2021-06-26:
Everytime I update the kernel, the module is rebuilded.
Is it really necessary?
Is there a way to disable this?

cilynx, 2021-06-26:
This is what DKMS does, by design.  It is not generally possible to use a module built for one version of the kernel with a different version of the kernel.  Yes it is really necessary.  Yes you can disable it by remove the DKMS scripts, however the next time you update the kernel, you will not get a version of the driver built that will work with your new kernel version.

All3xJ, 2021-07-03:
> This is what DKMS does, by design. It is not generally possible to use a module built for one version of the kernel with a different version of the kernel. Yes it is really necessary. Yes you can [...]

thanks for answer. so basically if I update to a new kernel there are much more possibilities that the rtl88x2bu driver will not work if I upggrade to a new kernel rather than it will work=

cilynx, 2021-07-03:
Yes, you are sadly correct.  This is exactly why this repository exists.  Neither @MaxG87 nor I nor any of the other awesome contributors actually wrote this driver.  Realtek puts out a new version of this driver every year or two and inevitably, something in the Linux kernel changes which breaks the driver long before Realtek is even going to think about releasing an update.  The community on this repo does our best to make the tweaks necessary to keep up with the kernel changes and keep the latest driver available here working with the latest release of the kernel.  Sometimes we fall a little behind (we're all volunteers with day jobs, school, families, etc.), but this is the goal we're aiming for and we tend to stay way ahead of Realtek =)

All3xJ, 2021-07-04:
thanks for everything :)

closed 2021-07-04 by All3xJ