cilynx/rtl88x2bu #161: Problems with dkms build -m rtl88x2bu -v

philygee, 2021-07-13:
Hello,
I'm trying to install this driver on a clean Fedora installation, but running into problems with the build command. This is what the terminal tells me when I run 

`sudo dkms build -m rtl88x2bu -v ${VER}`

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=5.12.14-300.fc34.x86_64 KVER=5.12.14-300.fc34.x86_64 src=/usr/src/rtl88x2bu-5.6.1........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.12.14-300.fc34.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
```
And if it can be useful, here is the make.log file:

```
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.12.14-300.fc34.x86_64 (x86_64)
Tue 13 Jul 2021 04:37:06 PM EDT
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.12.14-300.fc34.x86_64/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/kernels/5.12.14-300.fc34.x86_64'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
/usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c:3196:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3196 |                 else
      |                 ^~~~
/usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c:3199:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3199 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
/usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c: In function ‘rtw_parse_sta_vendor_ie_8812’:
/usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c:3227:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3227 |                         for(i=0; i<9;i++)
      |                         ^~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:60,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c:3229:33: note: in expansion of macro ‘RTW_INFO’
 3229 |                                 RTW_INFO("\n");
      |                                 ^~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_linux.o
/usr/src/rtl88x2bu-5.6.1/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/usr/src/rtl88x2bu-5.6.1/core/efuse/rtw_efuse.c:781:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  781 |                 if (i % 16 == 0)
      |                 ^~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:60,
                 from /usr/src/rtl88x2bu-5.6.1/core/efuse/rtw_efuse.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_debug.h:231:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  231 |         do {\
      |         ^~
/usr/src/rtl88x2bu-5.6.1/core/efuse/rtw_efuse.c:783:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  783 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_android.o
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c:358:76: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |                         if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                                            ^~~~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/recv_linux.c:358:76: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1860: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.12.14-300.fc34.x86_64'
make: *** [Makefile:2284: modules] Error 2
```

reidrivenburgh, 2021-07-13:
I can't check at the moment, but I was able to get a newer version build to work on the latest fedora.  Maybe give the 5.8.7.1 branch a try?

reidrivenburgh, 2021-07-14:
I can now check, and the 5.8.7.1 branch built cleanly for me and works great with my Cudy WU1300.

philygee, 2021-07-17:
Thanks for checking that! Some progress - with 5.8.7.1 I was able to build and install, but now the hiccup is with running modprobe. 

`modprobe: ERROR: could not insert '88x2bu': Key was rejected by service`

reidrivenburgh, 2021-07-17:
I ran the deploy.sh script and it worked fine (and is still working fine after a kernel upgrade/reboot).  My PC is very old, though.  It seems the issue you're seeing is with UEFI secure boot.  This page has some suggestions:

https://github.com/jlam55555/veikk-linux-driver/issues/3

Good luck!

philygee, 2021-07-18:
Thanks, I'll keep digging!

philygee, 2021-07-18:
That seems to have solved it! Driver was installed without issue. Thanks for the help!

closed 2021-07-18 by philygee

mhagnumdw, 2022-02-18:
> [jlam55555/veikk-linux-driver#3](https://github.com/jlam55555/veikk-linux-driver/issues/3)

It helped me.