cilynx/rtl88x2bu #163: recompile fails for my setup - after 5.11 kernel issue

githolgi, 2021-08-05:
Hi,

Recomplilation for new kernel 5.11 fails.

`
  CC [M]  /home/li/installs/rtl88x2bu/os_dep/osdep_service.o
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2205:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2205 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2205:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2205 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2205:10: note: each undeclared identifier is reported only once for each function it appears in
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2245:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2245 |    oldfs = get_fs();
      |            ^~~~~~
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2247:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2247 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c: In function ‘storeToFile’:
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2284:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2284 |    oldfs = get_fs();
      |            ^~~~~~
/home/li/installs/rtl88x2bu/os_dep/osdep_service.c:2286:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2286 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:287: /home/li/installs/rtl88x2bu/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1848: /home/li/installs/rtl88x2bu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-25-generic'
make: *** [Makefile:2284: modules] Error 2
`

gondilon, 2021-08-11:
I have this exact same issue. The driver actually stopped working for me after the kernel update.

githolgi, 2021-08-11:
BTW: you can also find the issue [at AskUbuntu.com](https://askubuntu.com/questions/1356084/realtek-wireless-adapter-not-working-after-update-to-kernel-5-11-0-25)

russelltadams, 2021-08-11:
Same issue on my Ubuntu 20 and 5.11 when recompiling.

lesstif, 2021-08-16:
@githolgi i'm  solved it using https://github.com/RinCat/RTL88x2BU-Linux-Driver repository.

cilynx, 2021-09-14:
Hrm -- I'm not able to reproduce on Ubuntu 20.04.3 LTS running 5.11 -- will have to poke at it a bit more.  @MaxG87 -- are you able to reproduce?
```
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ date
Mon 13 Sep 2021 06:16:24 PM PDT
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ cat /proc/version
Linux version 5.11.0-34-generic (buildd@lgw01-amd64-001) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ git branch
* 5.8.7.1_35809.20191129_COEX20191120-7777
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ git pull
Already up to date.
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ make clean
#make -C /lib/modules/5.11.0-34-generic/build M=/home/rcw/Downloads/rtl88x2bu clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko */*/*/*.ur-safe
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko */*/*.ur-safe
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions *.ur-safe
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ make -j8
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.11.0-34-generic/build M=/home/rcw/Downloads/rtl88x2bu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_cmd.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_security.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_io.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_debug.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_ioctl_query.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_ioctl_set.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_ieee80211.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_mlme.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_mlme_ext.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_mi.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_wlan_util.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_vht.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_pwrctrl.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_rf.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_chplan.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_recv.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_sta_mgt.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_ap.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/mesh/rtw_mesh.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_xmit.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_p2p.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_rson.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_tdls.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_br_ext.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_iol.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_sreset.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_btcoex_wifionly.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_btcoex.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_beamforming.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_odm.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_rm.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_rm_fsm.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_rm_util.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/efuse/rtw_efuse.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/os_dep/osdep_service.o
[... 101 lines cut ...]
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/platform/platform_ops.o
  CC [M]  /home/rcw/Downloads/rtl88x2bu/core/rtw_mp.o
  LD [M]  /home/rcw/Downloads/rtl88x2bu/88x2bu.o
  MODPOST /home/rcw/Downloads/rtl88x2bu/Module.symvers
  CC [M]  /home/rcw/Downloads/rtl88x2bu/88x2bu.mod.o
  LD [M]  /home/rcw/Downloads/rtl88x2bu/88x2bu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
rcw@ubuntu2004:~/Downloads/rtl88x2bu$ 
```

MaxG87, 2021-09-16:
I will have a look in it, but it might take some days. I am very busy at the moment.

MaxG87, 2021-09-20:
I build a virtual machine that hosts Ubuntu 20.04. It runs Kernel 5.11.34, as reported by `uname -r`. In that VM I can build revision 9f242ddc2ec4c5f4f8393a6e6496514b995b05bd without any issues.
I would be very glad if the reporter could provide the revision s:he was using or some other hints that help to reproduce the issue.

closed 2022-03-30 by MaxG87

simchanu29, 2022-04-12:
Instead of reopening an issue it seems I have the exact same issue.

OS : Ubuntu 20.04
Kernel : 5.13.0-39-generic
Using revision : `commit a1c53f43fb9995fbe3ad26567079d6384626d350 (HEAD -> 5.6.1_30362.20181109_COEX20180928-6a6a, origin/HEAD, origin/5.6.1_30362.20181109_COEX20180928-6a6a)`

Stacktrace using build.sh : 
```
$ cat /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.13.0-39-generic (x86_64)
mar. 12 avril 2022 15:51:40 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-39-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.13.0-39-generic »
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘isFileReadable’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2205 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2205 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2245:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2245 |    oldfs = get_fs();
      |            ^~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2247:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2247 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘storeToFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2284:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct <anonymous>’} from type ‘int’
 2284 |    oldfs = get_fs();
      |            ^~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2286:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2286 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281 : /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o] Erreur 1
make[2]: *** Attente des tâches non terminées....
make[1]: *** [Makefile:1879 : /var/lib/dkms/rtl88x2bu/5.6.1/build] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.13.0-39-generic »
make: *** [Makefile:2284 : modules] Erreur 2
```