cilynx/rtl88x2bu #166: Can not build on Raspberry Pi 4 w/ Arch Linux ARM, tries to access wrong folder.

CharmsyPone, 2021-08-14:
Been banging my head on the wall for a good long while trying to figure out why this isn't working, first going from the AUR, all th way to trying to build it myself.

I get to this step: `sudo dkms build -m rtl88x2bu -v ${VER}` right? And I get an error, telling me to check the log. This is the log:
```
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.10.52-7-ARCH (armv7l)
Wed Aug 11 14:46:11 UTC 2021
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.10.52-7-ARCH/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/lib/modules/5.10.52-7-ARCH/build'
Makefile:682: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.10.52-7-ARCH/build'
make: *** [Makefile:2284: modules] Error 2
```

So I go into `/usr/lib/modules/5.10.52-7-ARCH/build/arch`, and type ls. There is no "armv7l" folder. The folder is called "arm". This is why the compilation isn't working, because it's trying to access the wrong folder.

For the record I'm using the raspberry pi headers for arch linux (linux-raspberrypi-headers (armv7h)), if that's of any help. And I'm on the latest version of the kernel at the time of writing, 5.10.52-7-ARCH.

MaxG87, 2021-09-20:
Thank you for reporting the issue. This indeed looks serious.
1. Have you seen the new commit 9f242ddc2ec4c5f4f8393a6e6496514b995b05bd. Does it maybe resolve the issue?
2. I currently teach myself how to use QEMU to be able to reproduce and debug situations like yours. If you by any chance are familiar with that I could need some assistance on setting up a VM for your situation.

MaxG87, 2021-09-23:
I just tried to compile the driver on Raspberry Pi 4, RaspiOs 32bit, and got the same error as above. The culprit was that I forgot to set the switches correct.
After I ran

    sed -i 's/ARM_RPI = n/ARM_RPI = y/' Makefile
    sed -i 's/I386_PC = y/I386_PC = n/' Makefile

everything worked as expected.

Feel free to reopen if that did not solve the issue for you.

closed 2021-09-23 by MaxG87