cilynx/rtl88x2bu #167: error when running install.sh script

hopjopper, 2021-08-14:
trying to run the install.sh script get these errors

`      | 
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:2190:10: error: implicit declaration of function ‘get_ds’; did you mean ‘get_da’? [-Werror=implicit-function-declaration]
 2190 |   set_fs(get_ds());
      |          ^~~~~~
      |          get_da
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:2190:10: error: incompatible type for argument 1 of ‘set_fs’
 2190 |   set_fs(get_ds());
      |          ^~~~~~~~
      |          |
      |          int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/sched/task.h:11,
                 from ./include/linux/sched/signal.h:9,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/osdep_service.h:47,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/drv_types.h:27,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:2228:11: error: incompatible type for argument 1 of ‘set_fs’
 2228 |    set_fs(get_ds());
      |           ^~~~~~~~
      |           |
      |           int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/sched/task.h:11,
                 from ./include/linux/sched/signal.h:9,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/osdep_service.h:47,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/drv_types.h:27,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c: In function ‘storeToFile’:
/home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:2263:11: error: incompatible type for argument 1 of ‘set_fs’
 2263 |    set_fs(get_ds());
      |           ^~~~~~~~
      |           |
      |           int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/sched/task.h:11,
                 from ./include/linux/sched/signal.h:9,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/osdep_service.h:47,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/include/drv_types.h:27,
                 from /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:275: /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1719: /home/dev/Desktop/RTL88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/driver/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-26-generic'
make: *** [Makefile:1986: modules] Error 2
##################################################
Compile make driver error: 2
Please check error Mesg
##################################################
`
i have ran the install script and this issue exists still i dont know how to solve it any help would be greatly appreciated

cilynx, 2021-09-04:
Please try the 5.8.7.1_35809.20191129_COEX20191120-7777 branch and let us know how it goes.  5.3.1 is several upstream updates behind.

MaxG87, 2021-09-20:
The mention of `set_fs` and `get_fs` let me suspect that you use something that predates rev. 8b3babea2ccf874e0ce2e9ec1e1ca91efd8e826f . Please mention the precise revision in case you are using some recent one.

closed 2021-09-20 by MaxG87