cilynx/rtl88x2bu #170: Build fails on 64 bit raspberry pi

desultory, 2021-09-11:
I fixed it by changing these lines in the makefile:

`ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
                ARCH := arm64
        else
                ARCH := arm64
        endif`

cilynx, 2021-09-12:
I think I have a fix for this -- testing a few platforms to make sure before pushing it.

desultory, 2021-09-12:
I was able to fix it just by overriding it, I think it's just an issue with the wrong ARCH being used.  I never got these drivers to work quite right, I ended up using these: https://github.com/morrownr/88x2bu

cilynx, 2021-09-12:
You're correct that it's just the wrong arch being used, but I need to make sure it detects rpi aarch64 correctly as arm64 while still getting the other platforms right as well.  Tested on 64-bit Raspberry Pi 4, 32-bit Raspberry Pi 1B, and on x86 PC.

closed 2021-09-12 by cilynx

desultory, 2021-09-12:
Were you able to get full AC AP support with the drivers? I was only able to get N speeds at about 150mbps

commit 9f242ddc (2021-09-12): Rework architecture detection
Fixes #170