cilynx/rtl88x2bu #172: deploy.sh fails on Linux 5.13.0

martinleopold, 2021-09-30:
After updating kernel to 5.13, I've tried to reinstall/update the driver.

```OS Version: Pop!_OS 20.04 LTS (GNU/Linux 5.13.0-7614-generic x86_64)```

```./deploy.sh``` fails on ```dkms build``` with:

```
...Log begins after rsync output...
Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.1/source ->
                 /usr/src/rtl88x2bu-5.8.7.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=5.13.0-7614-generic src=/usr/src/rtl88x2bu-5.8.7.1...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.8.7.1 not found
Error! Bad return status for module build on kernel: 5.13.0-7614-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log for more information.
```

Here is ```cat /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log```:
```
Do 30 Sep 2021 11:05:21 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-7614-generic/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-7614-generic'
make[2]: *** No rule to make target '/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.o', needed by '/var/lib/dkms/rtl88x2bu/5.8.7.1/build/88x2bu.o'.  Stop.
make[1]: *** [Makefile:1859: /var/lib/dkms/rtl88x2bu/5.8.7.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-7614-generic'
make: *** [Makefile:2390: modules] Error 2
```

My make is quite rusty, but I suppose it's a path error or something, since a simple `make clean; make` seems to compile just fine.
Any help appreciated!

lyoshenka, 2021-09-30:
Try commit 9f242ddc2ec4c5f4f8393a6e6496514b995b05bd. I think this problem was introduced in in the last few days.

martinleopold, 2021-10-01:
Thanks @lyoshenka! I tried the last few commits, and it seems it's actually the latest commit (f2bccb9) that introduces the error for me. The one before that (e3339ad) builds just fine.

joshOberhaus, 2021-10-06:
Had the same issue on Linux 5.11.0, with the same resolution (rolling back one commit). Likely the --cvs-exclude is excluding the core directory.

MaxG87, 2021-10-10:
I am very sorry! I thought it was a no-brainer and did not give it a try! I just push a revert and the issue should be fixed now.

closed 2021-10-10 by MaxG87