cilynx/rtl88x2bu #181: How to configure a minimal yet compatible Kernel

MaxG87, 2021-11-10:
When I try to find the reason the rtl88x2bu driver cannot compile for a given Linux kernel, I usually find the breaking commit by a `git bisect`. I have a [helpful script](https://github.com/MaxG87/mainline-kernel-helpers) for that, but it still requires me to compile the Kernel repeatedly.

Right now I have to use my default Kernel configuration in order to get a Kernel that allows to build the Wifi driver against it. This is annoying because it means the script has to run hours instead of minutes because it has to compile so much irrelevant stuff.

I would like to configure the Kernel such that only whats strictly necessary is compiled. Maybe, less good but still a huge improvement, I would disable all sections that are obviously irrelevant, like graphics, sound, bluetooth, file systems, etc. Unfortunately, I am unable to find good documentation on that and was unable to figure out how to do so on my own.

Does anyone know how to configure a kernel appropriately? In the best case I can imagine I could run `make allnoconfig` and enable specific settings using `scripts/config` afterwards.

CGarces, 2021-11-10:
I don't understand your use case.

Why you need to compile the full kernel in a out of tree driver?

Just get the kernel headers from the distribution that you need to test, see https://github.com/cilynx/rtl88x2bu/blob/5.8.7.1_35809.20191129_COEX20191120-7777/.github/workflows/build.yml

MaxG87, 2021-11-10:
First, I didn't know about the possibility to use kernel headers. Thank you for mentioning that.

Second, I fear it won't help here. There are is no distribution that provides headers for Linux 5.16rc1 and even if there was, I probably couldn't use that to bisect to a breaking commit.

martadinata666, 2021-11-11:
Assuming 5.16rc1 mean you compile your kernel personally? Then you should be know that to properly build and install kernel, also `/path/to/linux/` usually `/usr/src/linux bla bla` doesn't mean literally `/path/to/linux` or just use distribution kernel and kernel headers.

MaxG87, 2021-11-11:
I am sorry, but I don't understand your answer.
Yes, I am compiling my kernels personally, so I can test whether the rtl88x2bu compiles for them. Please note that I do not have to boot the compiled kernel, I just need to have it compiled. That might be wasteful, but it works.

Now I am looking for something that is less wasteful, e.g. only creating the headers.

Why are you pointing me to `/usr/src/linux bla bla` and what should I do then?

CGarces, 2021-11-11:
> First, I didn't know about the possibility to use kernel headers. Thank you for mentioning that.
> 
> Second, I fear it won't help here. There are is no distribution that provides headers for Linux 5.16rc1 and even if there was, I probably couldn't use that to bisect to a breaking commit.

Look again at the proposed script, it fetch the .deb files from https://kernel.ubuntu.com/~kernel-ppa/mainline/

https://github.com/cilynx/rtl88x2bu/blob/5.8.7.1_35809.20191129_COEX20191120-7777/.github/workflows/build.yml#L39

You can get  5.16rc1 because not exist (yet) but any other rc is available for testing.

About the build...

`make ARCH=x86_64 CROSS_COMPILE= -C /path/to/linux/ M=/path/to/rtl88x2bu  modules`

If you use header files use the KVER option, it will calculate the correct paths instead of use -C
 https://github.com/cilynx/rtl88x2bu/blob/5.8.7.1_35809.20191129_COEX20191120-7777/.github/workflows/build.yml#L51

martadinata666, 2021-11-11:
And the `/lib/modules/bla3/build` will redirect to `/usr/src/linux-headers-bla3` If installed via DEB usualy the source will be there, if it manually compiled then the `-C` options will redirec to your custom linux kernel src path.

MaxG87, 2022-01-11:
I stumbled across the solution rather by chance. One has to run `make prepare` on a configured Kernel source.

    make oldconfig
    make prepare

This does the trick for me.

closed 2022-01-11 by MaxG87