cilynx/rtl88x2bu #183: system freezes instantaneously

dbolser, 2021-11-12:
The first time I hit `insmod 88x2bu.ko` it didn't freeze right away, and I was even able to connect to my router. However, when I tried to turn off the other NICs, it froze. Now it freezes as soon as I `insmod`.

What debugging information do you want?

```
$ uname -a
Linux yu-MS-7C09 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ inxi -F
System:    Host: yu-MS-7C09 Kernel: 5.11.0-40-generic x86_64 bits: 64 Console: tty 1 
           Distro: Ubuntu 20.04.3 LTS (Focal Fossa) 
Machine:   Type: Desktop System: XENTA product: MS-7C09 v: 1.0 serial: <superuser/root required> 
           Mobo: Micro-Star model: H310M PRO-VDH PLUS (MS-7C09) v: 1.0 serial: <superuser/root required> 
           UEFI: American Megatrends v: 1.30 date: 09/04/2019 
CPU:       Topology: Quad Core model: Intel Core i3-9100F bits: 64 type: MCP L2 cache: 6144 KiB 
           Speed: 800 MHz min/max: 800/4200 MHz Core speeds (MHz): 1: 800 2: 800 3: 800 4: 800 
Graphics:  Device-1: NVIDIA GK208B [GeForce GT 710] driver: nvidia v: 470.82.00 
           Display: server: X.org 1.20.11 driver: nvidia unloaded: fbdev,modesetting,nouveau,vesa tty: 163x24 
           Message: Advanced graphics data unavailable in console. Try -G --display 
Audio:     Device-1: Intel 200 Series PCH HD Audio driver: snd_hda_intel 
           Device-2: NVIDIA GK208 HDMI/DP Audio driver: snd_hda_intel 
           Sound Server: ALSA v: k5.11.0-40-generic 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet driver: r8169 
           IF: enp2s0 state: down mac: 00:d8:61:cc:30:4c 
           Device-2: Realtek RTL8192EE PCIe Wireless Network Adapter driver: rtl8192ee 
           IF: wlp3s0 state: up mac: 00:0c:43:01:d0:08 
           Device-3: TP-Link 802.11ac NIC type: USB driver: usb-network 
           Device-4: Ralink MT7601U Wireless Adapter type: USB driver: mt7601u 
           IF: wlx20e2060c507e state: up mac: 20:e2:06:0c:50:7e 
           IF-ID-1: br-9efc42c176bd state: down mac: 02:42:db:3e:69:2f 
           IF-ID-2: docker0 state: down mac: 02:42:86:35:8c:39 
Drives:    Local Storage: total: 447.13 GiB used: 44.00 GiB (9.8%) 
           ID-1: /dev/sda vendor: Maxtor model: Z1 SSD 480GB size: 447.13 GiB 
Partition: ID-1: / size: 438.62 GiB used: 44.00 GiB (10.0%) fs: ext4 dev: /dev/sda2 
Sensors:   System Temperatures: cpu: 28.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Info:      Processes: 286 Uptime: 11m Memory: 7.71 GiB used: 1.45 GiB (18.8%) Init: systemd runlevel: 5 Shell: bash 
           inxi: 3.0.38 
```

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1bcf:0005 Sunplus Innovation Technology Inc. Optical Mouse
Bus 001 Device 004: ID 0461:0010 Primax Electronics, Ltd HP PR1101U / Primax PMX-KPR1101U Keyboard
Bus 001 Device 003: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 002: ID 2357:012d TP-Link 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$ lspci
00:00.0 Host bridge: Intel Corporation 8th Gen Core 4-core Desktop Processor Host Bridge/DRAM Registers [Coffee Lake S] (rev 07)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller
00:14.2 Signal processing controller: Intel Corporation 200 Series PCH Thermal Subsystem
00:16.0 Communication controller: Intel Corporation 200 Series PCH CSME HECI #1
00:17.0 SATA controller: Intel Corporation 200 Series PCH SATA controller [AHCI mode]
00:1c.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #5 (rev f0)
00:1c.5 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #6 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device a2ca
00:1f.2 Memory controller: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller
00:1f.3 Audio device: Intel Corporation 200 Series PCH HD Audio
00:1f.4 SMBus: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller
01:00.0 VGA compatible controller: NVIDIA Corporation GK208B [GeForce GT 710] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GK208 HDMI/DP Audio Controller (rev a1)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8192EE PCIe Wireless Network Adapter
```

```
$ ifconfig # Before insmod
br-9efc42c176bd: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.18.0.1  netmask 255.255.0.0  broadcast 172.18.255.255
        ether 02:42:db:3e:69:2f  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:86:35:8c:39  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp2s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:d8:61:cc:30:4c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1837  bytes 144090 (144.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1837  bytes 144090 (144.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.103.55  netmask 255.255.255.0  broadcast 192.168.103.255
        inet6 fe80::e411:e0aa:7889:1b23  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:43:01:d0:08  txqueuelen 1000  (Ethernet)
        RX packets 420  bytes 153443 (153.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 572  bytes 95328 (95.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx20e2060c507e: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.212  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::498b:81fb:58c9:d2b5  prefixlen 64  scopeid 0x20<link>
        ether 20:e2:06:0c:50:7e  txqueuelen 1000  (Ethernet)
        RX packets 731  bytes 69377 (69.3 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 691  bytes 107154 (107.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Thank you! This works perfectly on another machine :-)

MaxG87, 2022-03-30:
I remember that I had instantaneous freezes a while back too. Nothing really helped, I wasn't even able to find out the offending commit.

I fear your only option is to try out different Linux kernel versions :-/.

dbolser, 2022-04-01:
I tried again and it worked!

closed 2022-04-01 by dbolser