cilynx/rtl88x2bu #186: Build fails on Debian 11, with gcc 10

ptsubin, 2021-11-27:
On debian, with gcc version gcc (Debian 10.2.1-6) 10.2.1 20210110, below lines in Makefile doesn't evaluate to be true 

```
GCC_VER_49 := $(shell echo `$(CC) -dumpversion | cut -f1-2 -d.` \>= 4.9 | bc )
ifeq ($(GCC_VER_49),1)
 EXTRA_CFLAGS += -Wno-date-time # Fix compile error && warning on gcc 4.9 and later
endif
```

Which results in
`/usr/src/rtl88x2bu-5.8.7.1/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]`

Commenting out the conditional expressions and keeping EXTRA_CFLAGS line allows the compilation to pass.

MaxG87, 2022-03-30:
I am developing this driver exclusively on Debian. I switched to Debian/testing only very recently. I never had any problems.

Can you give any further information?

ptsubin, 2022-03-31:
I am not sure what changed, but are you using gcc 10 ? 

Moving the line 
EXTRA_CFLAGS += -Wno-date-time # Fix compile error && warning on gcc 4.9 and later
Outside the block allowed me to compile it back then. I have moved to Ubuntu recently, so, I am not able to check this again at the moment.

martadinata666, 2022-03-31:
So this maybe because old code not patched yet. I tried to compile this with debian sid gcc11 with default branch `5.8.7.35809 bla2`.

MaxG87, 2022-03-31:
I switched to bookworm on 7th of March. Up to then I was using GCC 10.2.1-6. Now its GCC 11 something. This means that e.g. revision 95d792c889edf16bd1b919711ece13e8701ea494 was compiled on Debian 11 without problems.

Could that problem be due to non-amd64 platform?

Since the reporter switched to Ubuntu and is not affected anymore, I close it. If the reported problem still is an issue, feel free to open a new error report (with a bit more information, preferably) and mention me.

closed 2022-03-31 by MaxG87