cilynx/rtl88x2bu #191: insmod: ERROR: could not insert module 88x2bu.ko: Invalid module format

zemanlx, 2021-12-20:
Hello,

I was able to use your module successfully with previous kernels but after (I think 5.11.0-40) I can only build the module using `make clean && make` but I am having trouble loading it using 

```bash
$ uname -a
Linux buntu 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ sudo insmod 88x2bu.ko
insmod: ERROR: could not insert module 88x2bu.ko: Invalid module format
```

Would you mind to looks at this issue? I am happy to give you more logs if needed.

chrmeehansr, 2022-08-05:
Were you able to resolve your issue? Same error on Linux 5.18.0-kali5-amd64 x86_64

zemanlx, 2022-08-05:
@chrmeehansr No, I did not get it working :disappointed:

thatswhatshesaid27, 2024-02-19:
do "sudo make clean" and then "sudo make" and try again