cilynx/rtl88x2bu #194: Can not connect to wifi

fapinheiro, 2022-01-24:
Hi,
After driver installation, can not connect to wifi. Besides the password is ok, the network manager asks for password in loop.
I would appreciate any help.

My wifi usb adapter:
https://produto.mercadolivre.com.br/MLB-1660709828-adaptador-usb-wireless-dual-band-ac1200-5ghz-1200mbps-antena-_JM?matt_tool=31508429&matt_word=&matt_source=google&matt_campaign_id=14303413595&matt_ad_group_id=125984286637&matt_match_type=&matt_network=g&matt_device=c&matt_creative=539354956068&matt_keyword=&matt_ad_position=&matt_ad_type=pla&matt_merchant_id=505241090&matt_product_id=MLB1660709828&matt_product_partition_id=1457960502647&matt_target_id=pla-1457960502647&gclid=EAIaIQobChMI2czl5bDJ9QIVjYKRCh3cYAw4EAQYASABEgLIsfD_BwE

uname -a (output)
Linux kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux

dmesg and journal logs below.
[log.txt](https://github.com/cilynx/rtl88x2bu/files/7922261/log.txt)
[log3.txt](https://github.com/cilynx/rtl88x2bu/files/7922262/log3.txt)

quasar51893, 2022-04-04:
Did you ever find a solution to this? I am also having the same issue, and all I find is dead end threads.

kuflierl, 2022-04-11:
Same problem here on 5.13.0-39-generic #44~20.04.1-Ubuntu x86_64 in virtual-box and bare metal.
Also trying to remove the module will cause a catastrophic hang in the system

fapinheiro, 2022-04-11:
> Same problem here on 5.13.0-39-generic #44~20.04.1-Ubuntu x86_64 in virtual-box and bare metal. Also trying to remove the module will cause a catastrophic hang in the system UPDATE: I somehow got the module working in the vm by enabling the VirtualBox Extension Pack and enabling usb 3.0. (same kernel version as bare metal)

Many thanks. It really worked for me with a plus config. I added an extension pack enableding USB 2.0 (in my case because of my adapter is 2.0), and also enabled dhcp in `sudo vi /etc/network/interfaces`.

quasar51893, 2022-04-11:
> Many thanks. It really worked for me with a plus config. I added an extension pack enableding USB 2.0 (in my case because of my adapter is 2.0), and also enabled dhcp in `sudo vi [...]

Is this solution for inside VB VM or for a OS installed on main drive? Were you able to solve the issues by enabling 2.0 support and DHCP within the config files? Can you detail what you changed specifically?