cilynx/rtl88x2bu #203: Association failed on Ubuntu 20.04 (5.4.0-107-generic), TP-LINK Archer T4U v3

pitrou, 2022-04-15:
This driver is working very unreliably for me, I manage to connect once in a while but most attempts fail. Here are some logs:

```
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1277] Config: added 'ssid' value 'Freebox-1BADDE'
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1277] Config: added 'scan_ssid' value '1'
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1278] Config: added 'freq_list' value '2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484'
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1278] Config: added 'bgscan' value 'simple:30:-70:86400'
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1278] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1278] Config: added 'psk' value '<hidden>'
Apr 15 09:21:44 fsol NetworkManager[1303]: <warn>  [1650007304.1301] sup-iface[0x55eeabe073e0,wlxc006c3c019e1]: connection disconnected (reason -3)
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1302] device (wlxc006c3c019e1): supplicant interface state: associating -> disconnected
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1302] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: associating -> disconnected
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1401] device (wlxc006c3c019e1): supplicant interface state: disconnected -> scanning
Apr 15 09:21:44 fsol NetworkManager[1303]: <info>  [1650007304.1401] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: disconnected -> scanning
Apr 15 09:21:50 fsol wpa_supplicant[1411]: wlxc006c3c019e1: Trying to associate with 20:66:cf:cb:a7:9d (SSID='Freebox-1BADDE' freq=2462 MHz)
Apr 15 09:21:50 fsol NetworkManager[1303]: <info>  [1650007310.3597] device (wlxc006c3c019e1): supplicant interface state: scanning -> associating
Apr 15 09:21:50 fsol NetworkManager[1303]: <info>  [1650007310.3597] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: scanning -> associating
Apr 15 09:21:50 fsol wpa_supplicant[1411]: wlxc006c3c019e1: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2462 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2462 cf2=0
Apr 15 09:21:56 fsol wpa_supplicant[1411]: wlxc006c3c019e1: CTRL-EVENT-ASSOC-REJECT status_code=1
Apr 15 09:21:56 fsol wpa_supplicant[1411]: wlxc006c3c019e1: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Apr 15 09:21:56 fsol NetworkManager[1303]: <info>  [1650007316.7575] device (wlxc006c3c019e1): supplicant interface state: associating -> disconnected
Apr 15 09:21:56 fsol NetworkManager[1303]: <info>  [1650007316.7575] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: associating -> disconnected
Apr 15 09:21:56 fsol NetworkManager[1303]: <warn>  [1650007316.7576] sup-iface[0x55eeabe073e0,wlxc006c3c019e1]: connection disconnected (reason 1)
Apr 15 09:21:57 fsol NetworkManager[1303]: <info>  [1650007317.7644] device (wlxc006c3c019e1): supplicant interface state: disconnected -> scanning
Apr 15 09:21:57 fsol NetworkManager[1303]: <info>  [1650007317.7645] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: disconnected -> scanning
Apr 15 09:22:01 fsol wpa_supplicant[1411]: wlxc006c3c019e1: Trying to associate with 20:66:cf:cb:a7:9d (SSID='Freebox-1BADDE' freq=2462 MHz)
Apr 15 09:22:01 fsol NetworkManager[1303]: <info>  [1650007321.1330] device (wlxc006c3c019e1): supplicant interface state: scanning -> associating
Apr 15 09:22:01 fsol NetworkManager[1303]: <info>  [1650007321.1330] device (p2p-dev-wlxc006c3c019e1): supplicant management interface state: scanning -> associating
Apr 15 09:22:09 fsol NetworkManager[1303]: <warn>  [1650007329.5029] device (wlxc006c3c019e1): Activation: (wifi) association took too long
Apr 15 09:22:09 fsol NetworkManager[1303]: <info>  [1650007329.5030] device (wlxc006c3c019e1): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Apr 15 09:22:09 fsol NetworkManager[1303]: <info>  [1650007329.5034] sup-iface[0x55eeabe073e0,wlxc006c3c019e1]: wps: type pbc start...
Apr 15 09:22:09 fsol NetworkManager[1303]: <warn>  [1650007329.5046] device (wlxc006c3c019e1): Activation: (wifi) asking for new secrets
Apr 15 09:22:09 fsol NetworkManager[1303]: <info>  [1650007329.5063] device (wlxc006c3c019e1): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Apr 15 09:22:09 fsol NetworkManager[1303]: <info>  [1650007329.5068] device (wlxc006c3c019e1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Apr 15 09:22:23 fsol kernel: [ 1254.466825] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Apr 15 09:22:23 fsol wpa_supplicant[1411]: wlxc006c3c019e1: CTRL-EVENT-DISCONNECTED bssid=20:66:cf:cb:a7:9d reason=3 locally_generated=1
Apr 15 09:22:23 fsol NetworkManager[1303]: <info>  [1650007343.8507] device (wlxc006c3c019e1): Activation: (wifi) connection 'Freebox-1BADDE' has security, and secrets exist.  No new secrets needed.
Apr 15 09:22:23 fsol wpa_supplicant[1411]: wlxc006c3c019e1: CTRL-EVENT-ASSOC-REJECT status_code=1
```

martadinata666, 2022-04-15:
in my case this is because region issues, router wifi region should match ubuntu wifi region. Likely `US`

pitrou, 2022-04-15:
> in my case this is because region issues, router wifi region should match ubuntu wifi region. Likely `US`

I don't think this is the case for me, as I manage to connect sporadically (perhaps once every 10 attempts). It's just very unreliable.

MaxG87, 2022-08-25:
I cannot help here, I'm afraid. I just fix compilation errors. This seems to be an issue with the driver itself.

arisada, 2022-11-04:
Same problem, it never associates despite detecting the WiFi networks. It might be a region issue but I'd expect more overlap between US and EU.