cilynx/rtl88x2bu #205: Build failure in kali 5.16.0-kali7-amd64

burpheart, 2022-05-05:
I tried to build the driver on the latest version of kali (5.16.0-kali7-amd64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/root/rtl88x2bu  modules
make[1]: 进入目录“/usr/src/linux-headers-5.16.0-kali7-amd64”
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-19) 11.2.0
  You are using:           gcc-11 (Debian 11.3.0-1) 11.3.0
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
make[1]: 离开目录“/usr/src/linux-headers-5.16.0-kali7-amd64”

danigc12, 2022-05-21:
I tried manually creating the missing file but I think it didn´t work.
This is the missing script.
https://sources.debian.org/src/linux/5.16.11-1/scripts/pahole-flags.sh/

MaxG87, 2022-08-25:
Is this still an issue? If so, could you give me a link to an x86 image for the distribution you experience the issue with? I'd like to try this out in a VM.

closed 2022-11-04 by MaxG87