cilynx/rtl88x2bu #206: Merge Into Linux Kernel

TCROC, 2022-05-21:
Would it be possible to get this merged into the Linux kernel so that rtl88x2bu devices work out of the box?  I'm not knowledgeable on this subject so I may not know what I'm talking about here.  I just know a lot of my older USB network dongles work, but the new ones have been an issue.

MaxG87, 2022-05-23:
The Kernel hackers are a bit picky about onboarding new drivers. At the very least they require someone who promises to maintain it, and I know noone who would be comfortable with that.
Furthermore, including a driver to the kernel, together with a new configuration option, is a very complex task. I once tried but did not get far. For instance, to my knowledge we would have to get rid of all these `if KERNEL_VERSION` preprocessor statements. Thus, we would have to invest a significant amount of time with very uncertain returns.

If you know someone who could assist, I would be glad to hear that. After all, having that driver in the kernel would be great.

CGarces, 2022-05-24:
@TCROC Upstream a RTL driver not is impossible but take a lot of efforts. This driver use a common framework for several devices under Linux, Windows, and Android, so you should remove a lot of code to keep only the relevant to the specific device and the specific S.O. (Linux), and replace a lot of custom code with the equivalent functionality at linux kernel...
Take a look at 
https://github.com/hadess/rtl8723bs
You can see the complete history about the 805 commits to send the rtl8723bs driver to the [linux staging area](https://github.com/torvalds/linux/tree/master/drivers/staging/rtl8723bs).