cilynx/rtl88x2bu #210: Module not starting at boot time

Leuug, 2022-06-24:
I'm using Fedora 36 (kernel 5.18.5-200.fc36.x86_64) and have compiled the driver. It works fine, but even after running `deploy.sh`, the kernel module does not seems to start at boot time, so I have to run `insmod 88x2bu.ko`

Leuug, 2022-06-26:
Fixed the problem.

I had to create a key called `dkms.der` at `/root` using openssl and enroll that key. Then I uncommented the line `sign_tool="/etc/dkms/sign_helper.sh"` in the `/etc/dkms/framework.conf` file

After that you just need to run the `deploy.sh`script (remember to remove the installed module if you had already installed it with dkms)

closed 2022-06-26 by Leuug

carlossuburvan, 2023-02-17:
Hi,
I have two problems:
- can't find the sign_helper.sh script
- MOKutil doesn't accept the key upon reboot
For precision, I'm a complete noob... Thanks!

reisaraujo-miguel, 2023-02-17:
It seems DKMS updated, this should do it:

## Step-by-step:
Ensure you have **openssl** and **mokutil** installed, then run:
```
sudo -i
```
```
openssl req -utf8 -new -x509 \
-newkey rsa:2048 -keyout /var/lib/dkms/mok.key \
-outform DER -out /var/lib/dkms/mok.pub \
-noenc -days 36500
```
**OBS:** By default DKMS searches for a key pair called **mok.key** and **mok.pub** at `/var/lib/dkms/`. if you already have a key pair at `/root/`, I think renaming and moving them should be enough.

```
mokutil --import /var/lib/dkms/mok.pub 
```
Insert a passphrase two times, and then **reboot** to enroll that key.

If you run `dkms status` and you see that the `rtl88x2bu/<module-version>` (in my case `rtl88x2bu/5.8.7.1`) module is installed, run:
```
dkms remove rtl88x2bu/<module-version>
```
After that you can run the `deploy.sh` script.

**OBS:** you don't need to change anything in the `framework.conf` file

reisaraujo-miguel, 2023-02-17:
> Hi, I have two problems:
> 
>     * can't find the sign_helper.sh script
> 
>     * MOKutil doesn't accept the key upon reboot
> 
>     * After entereing MOK on reboot, ubuntu won't load.
>       I'm on ubuntu 20.04 LTS
>       For precision, I'm a complete noob... Thanks!

About Ubuntu not loading, I'm not sure what happened there. I think you can go into your BIOS and clear all the keys saved and then try again

carlossuburvan, 2023-02-17:
Thanks a lot! Ubuntu still doesn't load, I can only enter recovery mode to access the root shell prompt, so I did this:
- renaming the keys (dkms.key --> mok.key and dkms.der to mok.der)
- moving the key from /root/ to /var/lib/dkms/
- Import mok.der with mokutils (failed, already imported)
- delete mok.der
 -reboot
- re-import mok.der in mokutils
- reboot (this time MOK recognized it at boot)

still ubuntu HMI doesn't load (I only get a blinking "_")

reisaraujo-miguel, 2023-02-17:
This doesn't seem to be related to enrolling the key, but I'm not sure what happened. If you can enter recovery mode, I think this means you can get to grub.

Have you done anything else before trying to enroll the key the first time? Like a System Update or something? Have you tried booting on an older kernel to see if it works? Usually there is the option to boot on previous kernel versions on grub.