cilynx/rtl88x2bu #213: wifi stops in approx 5min

vponlby, 2022-08-01:
Driver works fine but my wifi connectivity(internet browsing) stops in approx 5 min, i am unable to even ping the wifi switch. Once i stop the connection and start again it starts working fine.
Is there any settings to limit the connectivity?
I tried with a mobile hotspot and i don't see any issue with it, but once i connect to the wifi switch (5Ghz) the issue happens very frequently, sometime i works for a longer duration other its almost 5min (approx).
Any suggestion on how should i troubleshoot such issues.
OS: Ubuntu 22.04 LTS
Wifi Adapter: TP link Archer T4U (AC 1300 - USB 2.0)

brain2xml, 2022-08-03:
same on xubuntu 22.04 LTS
Wifi Adapter: TP link Archer T4U (AC 1300 - USB 2.0)

vponlby, 2022-08-03:
For me the issue got resolved after updating the OS and few more updates whatever i found ( not sure what was the issues), now this seems to work fine i checked is for last 2 days.
(USB 3.0 - previously typo error)

closed 2022-08-03 by vponlby