cilynx/rtl88x2bu #215: Cant make build

Azrael1980, 2022-08-18:
DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 5.18.0-kali5-amd64 (x86_64)
Thu Aug 18 03:18:29 AM EDT 2022
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0
  You are using:           gcc-11 (Debian 11.3.0-5) 11.3.0
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_debug.o
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_debug.c: In function ‘dump_drv_version’:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_debug.c:45:69: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                                     ^
In file included from /usr/src/rtl88x2bu-5.8.7.1/include/drv_types.h:60,
                 from /usr/src/rtl88x2bu-5.8.7.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_debug.c:45:51: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                   ^~~~~~~~
/usr/src/rtl88x2bu-5.8.7.1/include/rtw_debug.h:254:46: note: in definition of macro ‘RTW_PRINT_SEL’
  254 |                         _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                                              ^~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.18.0-kali5-common/scripts/Makefile.build:293: /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_debug.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-kali5-common/Makefile:1858: /var/lib/dkms/rtl88x2bu/5.8.7.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
make: *** [Makefile:2391: modules] Error 2

Reelix, 2022-08-18:
> /bin/sh: 1: bc: not found

Run: `sudo apt install bc`

Azrael1980, 2022-08-18:
Thx. All work!

closed 2022-08-25 by MaxG87