cilynx/rtl88x2bu #216: Cannot build on kernel 5.19.2 (GCC 12.1.1)

RocKing1001, 2022-08-19:
I was trying to build the driver on kernel 5.19.2 and I was getting this error. This is strange because it worked fine yesterday before the update. I am also on GCC version 12.1 so maybe that is what is causing the issue?

Error for `dkms build -m rtl88x2bu -v ${VER}`
```
Sign command: /usr/lib/modules/5.19.2-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' KVER=5.19.2-arch1-1 src=/usr/src/rtl88x2bu-5.8.7.1...................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.2-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log for more information.
```
And here is the log file
```
# make.log
DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 5.19.2-arch1-1 (x86_64)
Fri Aug 19 08:21:28 AM CEST 2022
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.2-arch1-1/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Entering directory '/usr/lib/modules/5.19.2-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme_ext.o
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme_ext.c: In function ‘rtw_append_assoc_req_owe_ie’:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme_ext.c:8998:13: warning: the comparison will always evaluate as ‘true’ for the address of ‘owe_ie’ will never be NULL [-Waddress]
 8998 |         if (sec->owe_ie && sec->owe_ie_len > 0) {
      |             ^~~
In file included from /usr/src/rtl88x2bu-5.8.7.1/include/drv_types.h:75,
                 from /usr/src/rtl88x2bu-5.8.7.1/core/rtw_mlme_ext.c:17:
/usr/src/rtl88x2bu-5.8.7.1/include/rtw_security.h:167:12: note: ‘owe_ie’ declared here
  167 |         u8 owe_ie[MAX_OWE_IE_LEN];/* added in assoc req */
      |            ^~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_sta_mgt.o
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_sta_mgt.c:384:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  384 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /usr/src/rtl88x2bu-5.8.7.1/include/drv_types.h:103,
                 from /usr/src/rtl88x2bu-5.8.7.1/core/rtw_sta_mgt.c:17:
/usr/src/rtl88x2bu-5.8.7.1/include/sta_info.h:263:17: note: ‘lock’ declared here
  263 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ap.o
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_ap.c: In function ‘rtw_ap_update_chbw_by_ifbmp’:
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_ap.c:4403:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4403 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
In file included from /usr/src/rtl88x2bu-5.8.7.1/core/rtw_ap.c:17:
/usr/src/rtl88x2bu-5.8.7.1/include/drv_types.h:1124:19: note: ‘padapters’ declared here
 1124 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
/usr/src/rtl88x2bu-5.8.7.1/core/rtw_ap.c:4423:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4423 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
/usr/src/rtl88x2bu-5.8.7.1/include/drv_types.h:1124:19: note: ‘padapters’ declared here
 1124 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
[... 39 lines cut ...]
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c: In function ‘_cfg80211_check_bss’:
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:951:19: error: ‘struct wireless_dev’ has no member named ‘ssid_len’
  951 |         if ((!wdev->ssid_len) || (wdev->ssid_len != network->Ssid.SsidLength)
      |                   ^~
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:951:39: error: ‘struct wireless_dev’ has no member named ‘ssid_len’
  951 |         if ((!wdev->ssid_len) || (wdev->ssid_len != network->Ssid.SsidLength)
      |                                       ^~
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:952:33: error: ‘struct wireless_dev’ has no member named ‘ssid’
  952 |             || (_rtw_memcmp(wdev->ssid, network->Ssid.Ssid,
      |                                 ^~
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c: At top level:
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:10173:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10173 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:10173:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/usr/src/rtl88x2bu-5.8.7.1/os_dep/linux/ioctl_cfg80211.c:10456:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10456 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/rtl88x2bu/5.8.7.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1849: /var/lib/dkms/rtl88x2bu/5.8.7.1/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.2-arch1-1/build'
make: *** [Makefile:2391: modules] Error 2
```

CGarces, 2022-08-19:
Code needs to be patched for kernels >=5.19.2 and 6.0
Feel free to adapt this
https://github.com/Mange/rtl8192eu-linux-driver/pull/285
And send a PR.

liaten, 2022-08-19:
[image]
Still not working

liaten, 2022-08-19:
> [image] Still not working

[image]

liaten, 2022-08-19:
Tried [this](https://stackoverflow.com/questions/11561261/how-can-i-compile-without-warnings-being-treated-as-errors)
Still the same error

MaxG87, 2022-08-25:
What we see here is not a warning but a typing issue. Thus, disabling `-Werror` will not help here.

The issue is that some breaking changes from the 6.0 cycle were backported to 5.19. This is the first time this happened, so I was unaware of that possibility. I am sorry for that. Anyhow, in the fix-6.0 branch with commit 610b283b1e117afad686147d9b37992d5c4deb6e the issue shown above should be fixed.

However, it was noted that gcc 12.1 is used. This will provoke another issue, because gcc 12.1 will detect some new real issues. You could either create manually merge the fix-gcc12 branch onto fix-6.0 or simply apply the following patch with `git apply`.

I hope that this will allow you to compile the driver again.

```
diff --git a/Makefile b/Makefile
index db890fa..b06bbbe 100755
--- a/Makefile
+++ b/Makefile
@@ -11,6 +11,7 @@ EXTRA_CFLAGS += -Wno-unused-function
 EXTRA_CFLAGS += -Wno-unused-parameter
 EXTRA_CFLAGS += -Wno-unused-variable
 EXTRA_CFLAGS += -Wno-array-bounds
+EXTRA_CFLAGS += -Wno-address
 
 GCC_VER_49 := $(shell echo `$(CC) -dumpversion | cut -f1-2 -d.` \>= 4.9 | bc )
 ifeq ($(GCC_VER_49),1)
```

closed 2022-09-13 by MaxG87