cilynx/rtl88x2bu #217: Branch fix-6.0 unable to connect (Archer T4U)

Fjodor42, 2022-08-23:
With Linux 6.0.0rc2, the relevant branch compiles, to module loads etc.

Unfortunately, authentication with my AP times out:

From `/var/log/syslog`:

```
Aug 23 18:16:29 elli kernel: [  270.828103] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Aug 23 18:16:29 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-DISCONNECTED bssid=2c:56:dc:5e:ac:88 reason=3 locally_generated=1
Aug 23 18:16:29 elli wpa_supplicant[1878]: BSSID 2c:56:dc:5e:ac:88 ignore list count incremented to 2, ignoring for 10 seconds
Aug 23 18:16:29 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-SSID-TEMP-DISABLED id=1 ssid="foo 5GHz-2" auth_failures=2 duration=20 reason=CONN_FAILED
Aug 23 18:16:29 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-DSCP-POLICY clear_all
Aug 23 18:16:29 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Aug 23 18:16:35 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-SSID-REENABLED id=0 ssid="foo 5GHz-1"
Aug 23 18:16:35 elli wpa_supplicant[1878]: wlx60a4b753ab19: Trying to associate with 2c:56:dc:5e:ac:84 (SSID='foo 5GHz-1' freq=5220 MHz)
Aug 23 18:16:36 elli kernel: [  277.665475] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Aug 23 18:16:36 elli wpa_supplicant[1878]: wlx60a4b753ab19: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5220 ht_enabled=1 ch_offset=1 ch_width=40 MHz cf1=5230 cf2=0
Aug 23 18:16:46 elli wpa_supplicant[1878]: wlx60a4b753ab19: Authentication with 2c:56:dc:5e:ac:84 timed out.
```

MaxG87, 2022-08-25:
I am very sorry to hear that. Unfortunately I do not really know what the issue could be. Usually I only fix compilation errors.

Do you have any suggestion?

Fjodor42, 2022-08-25:
Not really, no, @MaxG87 - we seem to be of the same sort.

ironbone, 2022-08-26:
The same problem also in my case

MaxG87, 2022-08-29:
I went through the patches again, but I couldn't find any where I came up with an alternative fix. Furthermore, all occasions are just argument forwarding, nothing that directly affects a loop or so. So, regarding fixing the driver I really have no idea.

Could it be that your issue is that Arch started to ship RTW88? As described here: https://github.com/RinCat/RTL88x2BU-Linux-Driver/#linux-518-and-rtw88-driver?

Fjodor42, 2022-08-29:
Most relevant question, to be sure, bu no, it worked (and works) with 5.19.1.

MaxG87, 2022-08-31:
I compiled and booted 6.0-rc3 and I am affected too. I will try to narrow down the issue. The last time I did this it lead to nowhere, though. Let's hope this time bisecting a Kernel helps.

MaxG87, 2022-09-02:
I think that the issue manifests with Kernel v5.19.2 but not with v5.19.1. I had different results on a second computer, but I think this is because I skipped `make clean` before compiling the driver.

@Fjodor42 , @ironbone : Could you please test whether on your computers the regression becomes introduced between v5.19.1 and v5.19.2 too?

Fjodor42, 2022-09-02:
> @Fjodor42 , @ironbone : Could you please test whether on your computers the regression becomes introduced between v5.19.1 and v5.19.2 too?

Confirmed

kurtie, 2022-09-05:
Hi!

   Same here. I am on kernel v.5.19.4 (Fedora 36) and it compiles (default branch was not compiling since 5.19) but now it cannot connect to any WiFi. It says password is wrong every time. Tried also to make manually (make clean; make; insmod) instead of using dkms but it's just the same, module compiles, but then it cannot connect to any network.

MaxG87, 2022-09-05:
Thank you for your report. The error I get is exactly the same.I tried to narrow down the issue over the last days, but just when I thought I found the regression introducing commit, I learned I need to compile the kernel in a more expensive way (localmodconfig vs oldconfig). So it will take some more days to find it.Unfortunately I have to reboot for each compiled kernel. Is someone aware of a way to test a new kernel in a virtual machine?

MaxG87, 2022-09-11:
I reached out to the author of the offending commit. He pointed out a wrong line and that helped me to fix the connection issues.

I submit this comment running Kernel 5.19.6 from Debian. I would be glad if someone else could test the driver with other kernels, e.g. 6.0-rc4 or 5.19.8. I will close this issue as soon as some successful reports arrive.

Thank you very much for your patience.

kurtie, 2022-09-11:
Hi Max,

   Tested in FC 5.19.6 and with this fix now works. Updated to 5.19.8, and works like a charm.

   Thanks a lot.

marioalexis84, 2022-09-11:
On Arch Linux 5.19.7 kernel it works fine.

Fjodor42, 2022-09-12:
Confirmed working on 6.0-rc4 - thank you very much, @MaxG87

closed 2022-09-13 by MaxG87