cilynx/rtl88x2bu #218: Branch fix-6.0 also partially needed for 5.19.3

Fjodor42, 2022-08-23:
Discovering #217, I wanted to downgrade to 5.19.3.

Then I discovered that in order for the code to compile, I should change all instances of

`#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 0, 0)` to

`#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 3)`

**Except for** the one at line 1179 in `[os_dep/linux/ioctl_cfg80211.c](https://github.com/cilynx/rtl88x2bu/compare/5.8.7.1_35809.20191129_COEX20191120-7777...fix-6.0#diff-f42b6f2928f2ff74d519a664378a6267da1fcde5b9c08d67e46d462b07d5671b)`

Still the same symptoms as #217, though, so I am back on 5.19.1

Do note that I didn't try 5.19.2

CGarces, 2022-08-24:
Should be 
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 2)

MaxG87, 2022-08-25:
Fixed with 610b283b1e117afad686147d9b37992d5c4deb6e.

closed 2022-08-25 by MaxG87