cilynx/rtl88x2bu #221: prerequies ?

gritche70, 2022-09-27:
Hy, i try to compile this drivers since two day under debian 10 with GCC-10
but there lot, lot of problem like : 

In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:33,
                 from ./include/linux/if.h:28,
                 from ./include/linux/compat.h:17,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /usr/src/rtl88x2bu/include/osdep_service_linux.h:30,
                 from /usr/src/rtl88x2bu/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu/core/rtw_cmd.c:17:
./include/net/netlink.h: In function ‘nlmsg_unicast’:
./include/net/netlink.h:608:41: error: ‘MSG_DONTWAIT’ undeclared (first use in this function)
  608 |  err = netlink_unicast(sk, skb, portid, MSG_DONTWAIT);
      |                                         ^~~~~~~~~~~~
In file included from ./include/linux/kernel.h:10,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:9,
                 from /usr/src/rtl88x2bu/include/basic_types.h:76,
                 from /usr/src/rtl88x2bu/include/drv_types.h:26,
                 from /usr/src/rtl88x2bu/core/rtw_cmd.c:17:
./include/net/sock.h: In function ‘sk_peek_offset’:
./include/net/sock.h:512:23: error: ‘MSG_PEEK’ undeclared (first use in this function)
  512 |  if (unlikely(flags & MSG_PEEK)) {
      |                       ^~~~~~~~
./include/linux/compiler.h:77:42: note: in definition of macro ‘unlikely’
   77 | # define unlikely(x) __builtin_expect(!!(x), 0)
      |                                          ^
/usr/src/rtl88x2bu/core/rtw_cmd.c: In function ‘rtw_joinbss_cmd’:
/usr/src/rtl88x2bu/core/rtw_cmd.c:1501:39: warning: taking address of packed member of ‘struct _WLAN_BSSID_EX’ may result in an unaligned pointer value [-Waddress-of-packed-member]
 1501 |      pnetwork->network.IELength - 12, &psecnetwork->IELength,
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
/usr/src/rtl88x2bu/core/rtw_cmd.c:1515:32: warning: taking address of packed member of ‘struct _WLAN_BSSID_EX’ may result in an unaligned pointer value [-Waddress-of-packed-member]
 1515 |    pnetwork->network.IELength, &psecnetwork->IELength);
      |                                ^~~~~~~~~~~~~~~~~~~~~~
/usr/src/rtl88x2bu/core/rtw_cmd.c:1520:57: warning: taking address of packed member of ‘struct _WLAN_BSSID_EX’ may result in an unaligned pointer value [-Waddress-of-packed-member]
 1520 |  rtw_append_exented_cap(padapter, &psecnetwork->IEs[0], &psecnetwork->IELength);
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:327 : /usr/src/rtl88x2bu/core/rtw_cmd.o] Erreur 1
make[1]: *** [Makefile:1535 : _module_/usr/src/rtl88x2bu] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.14.110-mptcp »
make: *** [Makefile:2284 : modules] Erreur 2

so is there any requirement for being to sure to success ?

gritche70, 2022-09-28:
root@bastion:/usr/src/rtl88x2bu# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-18-amd64/build M=/usr/src/rtl88x2bu  modules                                                                                          
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.10.0-18-amd64 »
  CC [M]  /usr/src/rtl88x2bu/core/rtw_cmd.o
  CC [M]  /usr/src/rtl88x2bu/core/rtw_security.o
  CC [M]  /usr/src/rtl88x2bu/core/rtw_debug.o
corrupted size vs. prev_size
Aborted
make[3]: *** [/usr/src/linux-headers-5.10.0-18-common/scripts/Makefile.build:291 : /usr/src/rtl88x2bu/core/rtw_debug.o] Erreur 134                                                           
make[2]: *** [/usr/src/linux-headers-5.10.0-18-common/Makefile:1856 : /usr/src/rtl88x2bu] Erreur 2                                                                                           
make[1]: *** [/usr/src/linux-headers-5.10.0-18-common/Makefile:185 : __sub-make] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.10.0-18-amd64 »
make: *** [Makefile:2391 : modules] Erreur 2

does anoyone compile this driver whith success on an 5.10.0-18 debian kernel ?

MaxG87, 2022-10-04:
I was able to compile for v5.10.146. I was using GCC 12, though. However, I do not see a reason why GCC 10 wouldn't work anymore.

closed 2024-10-08 by MaxG87