cilynx/rtl88x2bu #230: Working openwrt package

zz64, 2022-10-22:
Hi,
i have successfully built this driver on my gentoo host with latest kernel 5.19.9-gentoo.
But I can't find a package that compiles under **openwrt 22.03.2 x86_64**
```CC [M]  /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/os_dep/linux/ioctl_cfg80211.o
/home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/os_dep/linux/ioctl_cfg80211.c: In function 'rtw_cfg80211_ch_switch_notify':
/home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/os_dep/linux/ioctl_cfg80211.c:441:17: error: too many arguments to function 'cfg80211_ch_switch_started_notify'
  441 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/include/osdep_service_linux.h:94,
                 from /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/include/osdep_service.h:53,
                 from /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/include/drv_types.h:27,
                 from /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7525:6: note: declared here
 7525 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:286: /home/moi/src/openwrt-sdk-22.03.2-x86-64_gcc-11.2.0_musl.Linux-x86_64/build_dir/target-x86_64_musl/linux-x86_64/rtl88x2bu-2022-03-25-1751f39d/os_dep/linux/ioctl_cfg80211.o] Error 1
```

using: 
https://github.com/hkucs-makerlab/openwrt-rtl8812bu-package/tree/openwrt-22.03
https://gitlab.com/_jason/openwrt-rtl8812bu-package

any working openwrt package ?

Thanks.

MaxG87, 2022-11-04:
This looks very much like an issue I fixed for Kernel v6.0. What kind of kernel does OpenWRT 22.03.2 run? Do you use the most recent version of this driver?

closed 2023-01-05 by MaxG87

t3therdev, 2023-01-06:
It runs 5.10.146, were you able to get it working on 22.02? or 22.03?