cilynx/rtl88x2bu #231: Your kernel headers for kernel 5.17.0-1020-oem cannot be found.

NayKeys, 2022-10-25:
Hi I am trying to install this driver on my machine but I encountered this error during dkms build and I don't know how to solve it. Thanks!

Error! Your kernel headers for kernel 5.17.0-1020-oem cannot be found.
Please install the linux-headers-5.17.0-1020-oem package or use the --kernelsourcedir option to tell DKMS where it's located.
nyasoftware@nyasoftware-System-Product-Name:~/

nye0, 2022-10-28:
I have the similar issue in raspberrypi. 
Just solved by
```
sudo dkms remove rtl88x2bu/${VER} --all
rm -rf /usr/src/rtl88x2bu-${VER}
sudo apt install --reinstall raspberrypi-kernel raspberrypi-kernel-headers
```
I think reason may because the kernel is not updated to the last one.
[ref](https://github.com/aircrack-ng/rtl8812au/issues/717)

closed 2023-01-05 by MaxG87

onlize, 2023-06-19:
I have the following error message:
`pi@raspberrypi:~ $ ls
rtl8812au
pi@raspberrypi:~ $ sudo dkms remove rtl88x2bu/${VER} --all
Error! Arguments <module> and <module-version> are not specified
Usage: remove <module>/<module-version> or
       remove -m <module>/<module-version> or
       remove -m <module> -v <module-version>
pi@raspberrypi:~ $
`

MaxG87, 2023-06-20:
Hi @onlize , have you set the `${VER}` shell variable? Your snippet lets me suspect it is unset. In order to remove all driver modules, I use `sudo dkms remove -m rtl88x2bu -v 5.8.7.1 --all`, which works without any variables. Since the version virtually never changes, there is no need to keep the snippet flexible.

MaxG87, 2023-06-20:
While here, I need to add that the above mentioned trick does not work for me. I am on Raspbian, with Kernel `6.1.21-v8+` and albeit I ran `sudo apt install --reinstall raspberrypi-kernel raspberrypi-kernel-headers` I still cannot compile the driver module!