cilynx/rtl88x2bu #240: Compiles too clean.

zaphida, 2023-02-11:
Literally no errors and only 1 warning on the first try and no warnings on the 2nd try... it's rather unsettling. But great!

rolex905, 2023-02-11:
Having some glitch where I cannot unsubscribe from this thread. Or rather, I have, but keep getting emails about it. Is this a GitHub glitch?

cilynx, 2023-02-11:
@zaphida -- glad it's working for you!

@rolex905 -- I just checked and you're watching the repo.  You can adjust your watch to only notify you on mentions or threads you comment on, or you can unwatch entirely to no longer be notified of any activity in the repo.

rolex905, 2023-02-12:
Thank you! I just turned off the alerts - I'm not a developer so I wasn't sure how to disable them. Thank you for your help!