cilynx/rtl88x2bu #245: Fails to work on Ubuntu 22.04.2 (kernel 5.19.0)

david-sitsky, 2023-05-04:
This doesn't seem to work on this setup.  Any ideas what might be wrong?

```
sits@diannao9:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
sits@diannao9:~$ uname -a
Linux diannao9 5.19.0-41-generic #42~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 18 17:40:00 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

```
May  4 10:06:11 diannao9 kernel: [  215.462840] usb 1-6: new high-speed USB device number 9 using xhci_hcd
May  4 10:06:11 diannao9 kernel: [  215.611286] usb 1-6: New USB device found, idVendor=2001, idProduct=331c, bcdDevice= 2.10
May  4 10:06:11 diannao9 kernel: [  215.611296] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May  4 10:06:11 diannao9 kernel: [  215.611300] usb 1-6: Product: 802.11ac NIC
May  4 10:06:11 diannao9 kernel: [  215.611303] usb 1-6: SerialNumber: 123456
May  4 10:06:11 diannao9 kernel: [  215.614801] RTW: [HALMAC]55772M
May  4 10:06:11 diannao9 kernel: [  215.614801] HALMAC_MAJOR_VER = 1
May  4 10:06:11 diannao9 kernel: [  215.614801] HALMAC_PROTOTYPE_VER = 6
May  4 10:06:11 diannao9 kernel: [  215.614801] HALMAC_MINOR_VER = 6
May  4 10:06:11 diannao9 kernel: [  215.614801] HALMAC_PATCH_VER = 23
May  4 10:06:11 diannao9 kernel: [  215.615557] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
May  4 10:06:11 diannao9 kernel: [  215.779761] RTW: HW EFUSE
May  4 10:06:11 diannao9 kernel: [  215.779770] RTW: 0x000: 29 81 00 3C  09 00 B1 00  94 07 64 00  00 00 A3 00  
May  4 10:06:11 diannao9 kernel: [  215.779792] RTW: 0x010: 28 29 2A 2A  2A 2A 25 26  27 27 27 02  00 00 FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779813] RTW: 0x020: FF FF 20 1F  1E 1D 1A 18  17 15 18 1B  1F 1F 1F 1F  
May  4 10:06:11 diannao9 kernel: [  215.779834] RTW: 0x030: 02 00 FF FF  00 FF 2C 00  FF FF 2B 2C  2C 2C 2C 2C  
May  4 10:06:11 diannao9 kernel: [  215.779854] RTW: 0x040: 28 29 29 29  29 22 00 00  FF FF FF FF  17 16 15 14  
May  4 10:06:11 diannao9 kernel: [  215.779874] RTW: 0x050: 14 13 12 11  13 15 18 18  17 17 01 00  FF FF 00 FF  
May  4 10:06:11 diannao9 kernel: [  215.779894] RTW: 0x060: 2C 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779914] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779934] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779954] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779974] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.779994] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  35 3F 1C 00  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780014] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 02 FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780035] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780055] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780075] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780095] RTW: 0x100: 01 20 1C 33  C1 67 02 C4  E9 0A 2F F6  76 02 03 0E  
May  4 10:06:11 diannao9 kernel: [  215.780116] RTW: 0x110: 03 38 30 32  2E 31 31 61  63 20 4E 49  43 08 03 31  
May  4 10:06:11 diannao9 kernel: [  215.780136] RTW: 0x120: 32 33 34 35  36 FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780156] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780176] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780196] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780216] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780236] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780256] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780276] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780296] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780317] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780337] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780357] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780377] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780397] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780417] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780437] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780458] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780478] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780498] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780518] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780538] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780558] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780578] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780598] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780618] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
May  4 10:06:11 diannao9 kernel: [  215.780638] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... 35 lines cut ...]
May  4 10:06:11 diannao9 NetworkManager[1023]: <info>  [1683158771.3929] rfkill2: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/ieee80211/phy2/rfkill2) (driver rtl88x2bu)
May  4 10:06:11 diannao9 kernel: [  215.853285] rtl88x2bu 1-6:1.0 wlxc4e90a2ff676: renamed from wlan0
May  4 10:06:11 diannao9 NetworkManager[1023]: <info>  [1683158771.4210] device (wlan0): interface index 5 renamed iface from 'wlan0' to 'wlxc4e90a2ff676'
May  4 10:06:11 diannao9 NetworkManager[1023]: <info>  [1683158771.4297] device (wlxc4e90a2ff676): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
May  4 10:06:11 diannao9 kernel: [  215.895238] RTW: [HALMAC][ALWAYS]shall R reg twice!!
May  4 10:06:11 diannao9 mtp-probe: checking bus 1, device 9: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6"
May  4 10:06:11 diannao9 mtp-probe: bus: 1, device: 9 was not an MTP device
May  4 10:06:11 diannao9 systemd-udevd[3771]: Using default interface naming scheme 'v249'.
May  4 10:06:11 diannao9 colord-sane: io/hpmud/musb.c 2101: Invalid usb_open: Permission denied
May  4 10:06:12 diannao9 kernel: [  216.550417] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
May  4 10:06:11 diannao9 colord-sane: message repeated 7 times: [ io/hpmud/musb.c 2101: Invalid usb_open: Permission denied]
May  4 10:06:12 diannao9 NetworkManager[1023]: <info>  [1683158772.1668] device (wlxc4e90a2ff676): supplicant interface state: internal-starting -> disconnected
May  4 10:06:12 diannao9 NetworkManager[1023]: <info>  [1683158772.1669] Wi-Fi P2P device controlled by interface wlxc4e90a2ff676 created
May  4 10:06:12 diannao9 NetworkManager[1023]: <info>  [1683158772.1674] manager: (p2p-dev-wlxc4e90a2ff676): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/7)
May  4 10:06:12 diannao9 NetworkManager[1023]: <info>  [1683158772.1677] device (p2p-dev-wlxc4e90a2ff676): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
[... cut ...]
May  4 10:06:15 diannao9 systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
May  4 10:06:16 diannao9 kernel: [  220.785205] RTW: nolinked power save enter
May  4 10:06:16 diannao9 systemd[1]: systemd-rfkill.service: Deactivated successfully.
May  4 10:06:16 diannao9 kernel: [  220.990139] RTW: [HALMAC][ALWAYS]shall R reg twice!!
May  4 10:06:17 diannao9 kernel: [  221.646611] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
May  4 10:06:17 diannao9 kernel: [  221.657573] RTW: nolinked power save leave
```

david-sitsky, 2023-05-05:
Perhaps I had a "stale" module loaded previously, but it seems to be working now.

MaxG87, 2023-05-18:
I am also using the driver under 22.04 successfully.

closed 2023-05-18 by MaxG87