cilynx/rtl88x2bu #251: DKMS binary package for rtl88x2bu: 5.13.1 not found

Ange7, 2023-06-01:
Hello,
I have this error when i run : `$ dkms autoinstall -m rtl88x2bu`

`
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
'make' -j16 KVER=5.19.0-41-generic KSRC=/lib/modules/5.19.0-41-generic/build......(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.19.0-41-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
`

How to install 5.13.1 ?

WilhelmFreiheitsberg, 2023-06-18:
Same issue here:

`$ sudo dkms autoinstall -m rtl88x2bu`

```
Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.19.0-45-generic KVER=5.19.0-45-generic src=/usr/src/rtl88x2bu-5.3.1..............(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.3.1 not found
Error! Bad return status for module build on kernel: 5.19.0-45-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log for more information.
```
When I open the make.log file:

```
DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.19.0-45-generic (x86_64)
dom 18 jun 2023 11:09:56 -03
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-45-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-45-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c:1828:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1828 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c:1830:9: note: here
 1830 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sreset.o
/usr/src/rtl88x2bu-5.3.1/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:257: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1857: /var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-45-generic'
make: *** [Makefile:2011: modules] Error 2
```

Any idea?

MaxG87, 2023-11-02:
I would be surprised if it could be installed using `dkms` directly. I never did so. Instead, there is the `deploy.sh` which works reasonably well. If there are issues with the deploy script, feel free to open a new issue.

closed 2023-11-02 by MaxG87