cilynx/rtl88x2bu #252: Kernel support for rtw88_8822bu

rrafal1337, 2023-06-06:
Hi. It seems that latest kernel 6.3.x supports rtl8822bu chips natively (there is no longer need to compile own driver in my case):

```
[Wed Jun  7 03:38:11 2023] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[Wed Jun  7 03:38:11 2023] usb 1-2: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[Wed Jun  7 03:38:11 2023] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Jun  7 03:38:11 2023] usb 1-2: Product: 802.11ac NIC
[Wed Jun  7 03:38:11 2023] usb 1-2: Manufacturer: Realtek
[Wed Jun  7 03:38:11 2023] usb 1-2: SerialNumber: 123456
[Wed Jun  7 03:38:11 2023] rtw_8822bu 1-2:1.0: Firmware version 27.2.0, H2C version 13
[Wed Jun  7 03:38:12 2023] usbcore: registered new interface driver rtw_8822bu
[Wed Jun  7 03:38:12 2023] rtw_8822bu 1-2:1.0 wlp4s0f3u2: renamed from wlan0
```
```
rru04@seagull ~ 12:04:50 $ modinfo rtw88_8822bu
filename:       /lib/modules/6.3.5-200.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822bu.ko.xz
license:        Dual BSD/GPL
description:    Realtek 802.11ac wireless 8822bu driver
author:         Realtek Corporation
alias:          usb:v20F4p808Ad*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v04CAp8602d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0E66p0025d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0846p9055d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0117d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0116d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p012Ed*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0115d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0138d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p012Dd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v13B1p0045d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v13B1p0043d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2001p331Fd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2001p331Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2001p331Ed*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p1874d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p1870d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p19AAd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p184Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p1841d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB81Ad*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pF822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pE822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pD822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pC822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pB822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp2102d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB82Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB812d*dc*dsc*dp*icFFiscFFipFFin*
depends:        rtw88_usb,rtw88_8822b
retpoline:      Y
intree:         Y
name:           rtw88_8822bu
vermagic:       6.3.5-200.fc38.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        76:77:D2:CA:8C:0A:B7:A7:39:59:A2:B0:7A:5A:E2:61:10:CB:E7:D0
sig_hashalgo:   sha256
signature:      78:13:CF:3B:DA:A9:52:99:FF:FE:F8:20:31:2D:08:47:B3:2A:D4:BA:
		7C:44:D4:0F:22:8E:C1:68:39:F8:EB:5B:28:B8:50:F3:B4:7D:E5:D0:
		D8:24:DE:65:8A:C8:A7:8D:CE:54:4E:51:C9:DB:F0:10:3B:36:45:D4:
		0B:F5:DB:42:3A:B6:EF:03:A9:9B:6D:2B:CC:65:99:03:52:A7:A5:A1:
		45:C4:B9:88:7F:FE:93:DE:44:21:33:4F:F7:F3:D4:FF:30:20:06:CA:
		31:15:EE:D6:86:95:D7:EF:31:3F:6C:8A:E2:F4:0D:31:11:83:1E:37:
		12:C2:B7:1F:98:B6:43:21:9E:9A:7E:C9:62:37:EE:63:9E:06:22:E0:
		F8:16:7E:39:AF:6E:C2:EE:96:56:62:63:F0:30:2E:E2:82:79:24:6B:
		A8:E0:B1:D0:84:F9:C6:BC:8F:87:92:C4:E0:57:CD:2F:AE:BC:C5:5A:
		96:E3:E7:81:E9:AC:23:F4:52:2F:26:E5:6E:6F:4D:48:5A:65:CB:14:
		7D:70:E5:CE:C7:32:35:0A:99:8F:06:2B:21:D7:3B:77:90:D6:2D:48:
		40:BE:29:E8:62:44:8C:0E:29:41:59:A7:F7:C1:C8:1D:79:CC:99:32:
		7C:4A:53:7E:19:FC:F2:A0:27:87:64:5E:C3:96:6A:42:5C:BB:83:96:
		1E:7A:87:BD:FC:29:B5:1D:4D:02:82:0B:07:06:91:79:88:81:D5:A7:
		AD:62:80:4A:2C:DF:1C:73:8B:5F:72:45:E6:E5:2E:C8:A0:20:1C:FA:
		20:DF:54:FD:86:7D:D0:4F:56:DA:54:FB:B3:E1:3B:98:81:EE:D2:F9:
		32:01:B4:F6:6F:FA:91:FD:A1:02:73:50:68:5E:A4:C9:E2:FB:25:FC:
		35:8C:C8:2E:CC:D5:1B:EC:68:ED:D6:D9:14:3B:8B:0B:6A:3D:23:D3:
		F5:10:2F:ED:08:25:32:3F:C0:32:1B:70:15:B7:95:D5:CF:78:5E:C4:
		18:DB:3F:8B:96:83:CC:16:A0:8D:BF:7A:DE:2F:F7:84:D5:AA:55:BC:
		CE:08:79:7B:2C:20:80:0D:B4:1E:E7:BE:8C:B3:67:15:A4:15:F8:22:
		8E:1C:85:DF:2E:1E:76:BC:39:43:80:B0:95:B3:54:51:DB:A2:EC:41:
		89:08:FF:75:64:F5:B2:C8:8F:2B:50:25:50:5C:3E:D7:1F:67:72:D6:
		C0:33:C4:17:AA:35:CE:32:5A:B3:B4:16:74:6E:D7:21:E8:11:32:6E:
		B0:15:38:F0:25:C9:C7:80:CE:70:5A:0D:E8:BC:75:DF:F9:2B:69:A2:
		B4:2E:3D:CB:7A:3B:F1:52:5E:52:31:FD
```

sebma, 2023-07-01:
@rrafal1337 Hi, Just to make this clear : 

Are you using Linux Fedora FC38 with kernel version 6.3.5 ?

Maybe you can change the title of this issue to more clear like `rtl8822bu chips are already supported by kernel v6.3.x`.

rrafal1337, 2023-07-01:
> @rrafal1337 Hi, Just to make this clear :
> 
> Are you using Linux Fedora FC38 with kernel version 6.3.5 ?

Yes. I was used previously driver from this project but now it seems that is included with kernel 6.3 in latest Fedora which solves my problem.

> 
> Maybe you can change the title of this issue to more clear like `rtl8822bu chips are already supported by kernel v6.3.x`.

Already changed. Thanks for suggestion.

schauveau, 2023-07-03:
The driver is not in all v6.3.x kernels.  I am currently running 6.3.0 in Debian Testing and it is not there.

According to https://packages.debian.org,  the module `rtw88_8822bu.ko` is in the linux-image-6.4.0 packages from Debian Experimental.

shenlebantongying, 2023-09-01:
I just came to confirm that Linux 6.4 (archlinux), RTL88x2bu works out of the box.

sebma, 2023-09-03:
@schauveau And also in Debian Sid :smile: : https://packages.debian.org/search?suite=sid&arch=amd64&mode=path&searchon=contents&keywords=rtw88_8822bu.ko

It is also in Ubuntu 23.04+ with kernel v6.2.0-20-generic : https://packages.ubuntu.com/search?suite=lunar&arch=amd64&mode=exactfilename&searchon=contents&keywords=rtw88_8822bu.ko .
And in Ubuntu mantic with kernel v6.3.0-7-generic : https://packages.ubuntu.com/search?suite=mantic&arch=amd64&mode=exactfilename&searchon=contents&keywords=rtw88_8822bu.ko

gittrekt, 2024-06-06:
Sorry for bumping old thread, but...

Linux fedora 6.8.11-300.fc40.x86_64
rtl8822bu works out of box, but crashes under heavy load. This is still a viable solution for now

MaxG87, 2024-10-08:
I just want to point out that for some users the driver appears to work at first glance but causes subtle issues later on. One particular problem is a loss of connectivity after about an hour of usage. More  subtle issues are reported [here](https://github.com/cilynx/rtl88x2bu/issues/270).

3vi1, 2024-10-08:
My experience matches @gittrekt.  I have two USB 88x2bu adapters and both of them will fail such that I need to unload/reload the drivers to get them going after just a few days.  I've seen this since they were introduced all the way to the current 6.11 mainline kernel.  I especially see the problems start showing when I'm doing system backups to my external USB drives.

@MaxG87 's drivers, on the other hand, are rock solid and never drop out - no matter the system load.

schauveau, 2024-10-08:
I also noticed that my USB `rtw88_8822bu` from kernel 6.10.6-amd64 (debian) is quite unstable. 
I occasionally have to unplug/re-plug it.
I never bothered to really investigate since this is a old machine that I intend to replace soon.