cilynx/rtl88x2bu #268: Error while compiling in `6.9.1` kernel

p0ryae, 2024-05-19:
Attempting to compile (or even deploy) will lead to the following error:

```sh
/home/porya/rtl88x2bu/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/porya/rtl88x2bu/os_dep/linux/ioctl_cfg80211.c:441:17: error: too many arguments to function ‘cfg80211_ch_switch_started_notify’
  441 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/porya/rtl88x2bu/include/osdep_service_linux.h:94,
                 from /home/porya/rtl88x2bu/include/osdep_service.h:53,
                 from /home/porya/rtl88x2bu/include/drv_types.h:27,
                 from /home/porya/rtl88x2bu/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8811:6: note: declared here
 8811 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/porya/rtl88x2bu/os_dep/linux/ioctl_cfg80211.c:457:9: error: too many arguments to function ‘cfg80211_ch_switch_notify’
  457 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8795:6: note: declared here
 8795 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /home/porya/rtl88x2bu/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/lib/modules/6.9.1-zen1-1-zen/build/Makefile:1917: /home/porya/rtl88x2bu] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.9.1-zen1-1-zen/build'
make: *** [Makefile:2395: modules] Error 2

```

I'm using Arch Linux, GCC verison is `14.1.1`

CGarces, 2024-05-19:
You need this patch
https://github.com/aircrack-ng/rtl8812au/pull/1149

MaxG87, 2024-05-21:
I just pushed a fix for compatibility with v6.9. I usually push using the newest driver and Kernel, which took some time this time due to the GCC 13 issues.

closed 2024-05-21 by MaxG87