cilynx/rtl88x2bu #270: Maintenance after Debian Trixie

MaxG87, 2024-05-21:
I recently migrated to Debian Trixie (which is still testing). Since then I have  the drivers `rtw88_8822b` and `rtw88_8822bu` available by the distribution. Therefore, I am no longer dog-footing the driver.

I think, soon the repository will not be required any more. I think it is reasonable to continue maintenance until the release of Debian Trixie. After that, I consider to stop fixing compilation issues.

Are there any thoughts on this? Should a remark on this be added to the README, so others can prepare or ask for longer support?

cilynx, 2024-05-22:
Agreed.  Readme has been updated with commentary on mainline and a link to this issue for anyone interested to comment.

S-K-Tiger, 2024-05-30:
For me when using rtw88_8822bu I found that, when attempting to connect to a network using saved credentials, it would ask for credentials and ask again if provided repeatedly. Installing rtl88x2bu has fixed this issue for me.

System info
```
Operating System: Kubuntu 24.04
KDE Plasma Version: 5.27.11
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.13
Kernel Version: 6.8.0-31-generic (64-bit)
[...]
System Version: 2.0
```
Device info
```
  idVendor           0x2357 TP-Link
  idProduct          0x0115 Archer T4U ver.3
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
```

3vi1, 2024-07-05:
Preface:  I have used your driver with Ubuntu and the latest mainline kernels with no problems for years.  Thanks so much for maintaining this!

I hope you can continue to maintain this for at least a short while longer.  Let me explain:

Since they added rtw88 support in the newer kernels, I've been trying to make that work instead.  While that "works" (at first), I'm of the opinion that the drivers in kernel are still suffering for major bugs that make them unsuitable for anything more than testing.

The main problem I experience with the default kernel drivers is that the interface goes down/disappears, usually about 24 hours into use (sometimes less, sometimes more), and the only way to re-establish it is to remove/re-insert the modules.  Dmesg points to "Failed to get tx report from firmware" or lps issues (though it happens when the wifi is being actively used).  I can expect it to happen anytime my systems been up for a day and I'm doing backups while downloading anything; somehow active USB disk activity seems to be a catalyst for the failures.

Eventually, reloading the drivers won't fix the issue and the system needs an entire reboot to get the wifi working... at which point it's good for maybe another day before it starts showing the same problems again.

I found others online complaining of similar issues.  After fighting with it for months and trying everything that other claimed had worked for them (tweaking kernel parameters, turning off wifi powersave, setting lps options on the drivers, etc.) I gave up and tried the lwfinger rtw88 drivers (just slightly newer versions of what's packaged with new kernels, from what I understand).  Unfortunately, no change in symptoms when using those.

I even bought a different USB dongle (same chipset) so that I could verify it wasn't a hardware issue.  It show the exact same issue.

After retrying all combination of settings, I finally decided to install your drivers again - to verify it really was the kernel drivers and not something else in my ever-changing system configuration.  **I've now been up for over 3 days without any issue at all**, I have never been able to go more than two full days using the drivers in the newer kernels.

So, again, I'm hoping you continue to keep these compiling with newer kernels until they shake out the issues in the mainlined drivers.  I'll continue testing them and report these issues upstream as newer kernels become available, and hopefully they get more feedback as other distros roll with newer kernels.

Thanks for all your hard work.  It has been appreciated.

MaxG87, 2024-07-23:
@S-K-Tiger , @3vi1: Thank you very much for sharing this.

I want to reassure you that I am not planning to stop maintenance on this driver in the near future. As stated above, I will continue maintenance at least until [Debian Trixie](https://www.debian.org/releases/trixie/) is released. After that, the driver will be functional for three additional releases, since Debian releases stick to the major kernel version the offered at release time.

I think it wouldn't be reasonable to use this driver much longer though. All I do is keeping it compatible with new Kernel releases. Only the in-tree driver will receive potential improvements or security fixes. If you experience issues with the in-tree driver, I want to encourage you report the bugs to your distribution or the maintainer directly.

3vi1, 2024-07-23:
That's great to hear.  I'll definitely be reporting issues whenever the amd64 6.10 kernel is available (it doesn't look like they've had a successful build since rc4 at https://kernel.ubuntu.com/mainline/v6.10/) and I can test.

Until then, I'm very grateful to have your drivers.  It's been 3 weeks now and yours have been steady as a rock whereas the in-kernel drivers all through 6.9 (maybe earlier... I forget) and 6.10rc4 fail after a day.

Lid2be, 2024-08-14:
Hi all. I've just started using Linux Mint 22 (6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux) and my WI-Fi adapter had not worked. I've compiled this project and run `sudo modprobe cfg80211` and `insmod 88x2bu.ko` and Wi-Fi works. What does it mean? Does the kernel I use have support included or not? TIA

MaxG87, 2024-08-19:
In my copy of the Kernel sources version 6.8 includes the folder `drivers/net/wireless/realtek/rtw88`. However, every distribution is free to provide and configure whatever it considers to be reasonable. So, maybe _your_ Kernel does not yet support your WIFI adapter. Since Linux Mint 22 is rather new, maybe tutorials on how to activate `trw_88_8822b` and `rtw88_8822bu` have yet to be written.

gn0, 2024-09-22:
@Lid2be On Ubuntu, it works for me after installing `linux-image-6.8.0-45-generic`  and (importantly) `linux-modules-extra-6.8.0-45-generic`. The extra kernel modules contain `rtw88_8822bu` which gets automatically loaded on boot if the adapter is plugged in.

jace-a-akerlund, 2024-09-28:
Thank-you @cilynx and @MaxG87 for maintaining this driver. 

I recently upgraded to Ubuntu 24.04.1 LTS and found that the drivers you have noted are installed for me by default, but they cease to work after about an hour. Additionally a generic speedtest between the rtw88 and 88x2bu drivers show that the 88x2bu sustains at least twice the bandwidth (on my system at least). I will periodically check the rtw88 driver for improvements, but will stick with your drivers for the moment.

Wifi Device:
```
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 RTL88x2bu [AC1200 Techkey]
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 USB3.0 802.11ac 1200M Adapter
```

MaxG87, 2024-10-08:
Thank you everyone for your responses. It means a lot to me to see that people depend on this.

Your messages helped me to understand better the connection loss I encountered. I wasn't aware that my issue was caused by the in-tree driver. I was accepting it without other thoughts. I switched back to the driver of the repository and will pay attention whether this helps.

I want to stress again that maintenance will endure at least until Trixie is released. Given the reported issues it is likely I will continue after it.

3vi1, 2024-10-08:
We really appreciate that @MaxG87 .  As of the 6.11 kernels, my testing shows the mainline drivers are still very flaky and lose connection after a couple of days (or even within hours sometimes).  I've tested with multiple external 88x2bu USB adapters and get the same results.

The drivers from this repo, however, continue to be rock solid with both adapters (in use 8 days at the moment with no issues).

DuAnto, 2024-12-24:
I am a holder of "(ID 2357:0115) TP-Link Archer T4U ver.3" WiFi dongle.  Approximately 1 year I was quite happy with mainline kernel.  But several days ago I've updated kernel and firmware in Manjaro Linux.  As a result of this action, mainline driver became unusable.  After some time it loses connection and everything hangs..  Inspection of log files showed:
[image]

In my Manjaro install, there are 3 kernels installed: 6.6, 6.11 and 6.12.  All of them have broken mainline driver.

Today I've installed this driver and had no troubles watching 2hr video using 6.12 kernel (while using mainline kernel I was unable even to clone this repo).

Thank you!

splitTh3byte, 2024-12-28:
Hi,
The ported version of the driver on the linux kernel has a major bug.
It gets disconnected after 5-10 minutes and i get this error : 
device (wlan0): link timed out.
device (wlan0): state change: activated -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
device (wlan0): supplicant interface state: scanning -> authenticating
device (wlan0): supplicant interface state: authenticating -> disconnected
My config : 
Debian 6.11.10+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.11.10-1~bpo12+1 (2024-12-19) x86_64 GNU/Linux

I blacklisted the built-in driver for the 6.11.10 linux kernel and loaded MaxG87's version. 
The issue has been solved. So, i'll continue to use Max's version. Much better and stable.
Thank you for the good work!

philenz, 2024-12-31:
Exactly same experience as previous commenters:

- Running Ubuntu 24.04.
- With ID 2357:0115 TP-Link Archer T4U ver.3.
- Drops connection frequently and won't reconnect unless I restart NetworkManager or reboot.
- And connection speed is not good.

So I installed your driver using ./deploy.sh, and did echo "blacklist rtw88_8822bu" | sudo tee /etc/modprobe.d/rtw8822bu.conf.

Now my Wifi device is working beautifully and at full speed :-)
Thank you so much!!!
Happy New Year.

dubhater, 2025-02-03:
> The main problem I experience with the default kernel drivers is that the interface goes down/disappears, usually about 24 hours into use (sometimes less, sometimes more), and the only way to [...]

I believe this is https://github.com/lwfinger/rtw88/issues/205. It's fixed in kernel 6.14.

Other things fixed recently:

* Repeated "USB disconnect" with some combinations of RTL88x2BU devices and USB controllers (I guess? mine didn't do it) (6.14)
* Bad performance with torrents and probably other things that use UDP (6.14)
* Supports 3 more USB IDs (6.14)
* The reported signal strength jumping to ~0 dBm / 100% when downloading something (not sure if this affected RTL88x2BU devices) (6.13)
* AP mode may actually work now (6.12)
* Better RX speeds, especially on some ARM boards (6.12)
* Automatic switching to USB 3 (6.12)
* Less likely to disconnect when there is beacon loss (6.11)
* The LED blinks now (6.14)

Please retest https://github.com/lwfinger/rtw88 and report any bugs over there. Or test 6.14-rc1 if you like.

MaxG87, 2025-02-09:
Thank you very much, @dubhater for your report. I will wait for Linux 6.14, give it a try and report back here.

I will close this ticket now. Great thanks to everyone reporting back here. I will wait for 6.14, see whether it will be included to Debian and whether it fixes the connection breaks. If relevant, I'll give a new announcement.

closed 2025-02-09 by MaxG87

3vi1, 2025-03-14:
Just noticed @dubhater 's message now that 6.14 is in the Ubuntu Plucky/devel repos and the current drivers don't compile for that kernel version.

I was always able to recreate the problem within a day or so.  I'll remove these drivers now, test for several days, and post back with the results.  Thanks!

3vi1, 2025-03-15:
Kernel drivers still appear to have problems.  Things seemed a little slow this morning, and then 26hrs since reboot they just stopped working completely until I rmmod/re-modprobed the rtw88_8822bu module.  :(

I'll collect more info an report to the lwfinger issues.

dubhater, 2026-03-05:
You can also disable the switch_usb_mode option:

```
echo 0 > /sys/module/rtw88_usb/parameters/switch_usb_mode
```

Then plug it again into a USB 3 port. It will stay in USB 2 mode.

3vi1, 2026-03-06:
> > Since I had problems building these with kernel 6.19
> 
> What problems exactly? It seems to build fine for me using the base linux source.

https://github.com/cilynx/rtl88x2bu/issues/284