cilynx/rtl88x2bu #276: Error! Bad return status for module build on kernel: 6.14.1-arch1-1 (x86_64)

PaulStogov, 2025-04-09:
Won't build against 6.14.1 on 64x ArchLinux. 
6.13.8 is ok

`(6/9) Install DKMS modules
==> dkms install --no-depmod vboxhost/7.1.4_OSE -k 6.14.1-arch1-1
==> dkms install --no-depmod rtl88x2bu/5.8.7.1 -k 6.14.1-arch1-1

Error! Bad return status for module build on kernel: 6.14.1-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtl88x2bu/5.8.7.1 -k 6.14.1-arch1-1' exited 10
==> dkms install --no-depmod acpi_call/1.2.2 -k 6.14.1-arch1-1
==> depmod 6.14.1-arch1-1
`

make.log is

`cat /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log
DKMS (dkms-3.1.7) make.log for rtl88x2bu/5.8.7.1 for kernel 6.14.1-arch1-1 (x86_64)
2025-04-09T18:07:40 MSK

Building module(s)
# command: 'make' -j 4 KVER=6.14.1-arch1-1 src=/usr/src/rtl88x2bu-5.8.7.1
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.14.1-arch1-1/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Entering directory '/usr/lib/modules/6.14.1-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/rtl88x2bu/5.8.7.1/build'
  CC [M]  core/rtw_cmd.o
  CC [M]  core/rtw_security.o
  CC [M]  core/rtw_debug.o
  CC [M]  core/rtw_io.o
  CC [M]  core/rtw_ioctl_query.o
  CC [M]  core/rtw_ioctl_set.o
  CC [M]  core/rtw_ieee80211.o
  CC [M]  core/rtw_mlme.o
  CC [M]  core/rtw_mlme_ext.o
  CC [M]  core/rtw_mi.o
  CC [M]  core/rtw_wlan_util.o
  CC [M]  core/rtw_vht.o
  CC [M]  core/rtw_pwrctrl.o
  CC [M]  core/rtw_rf.o
  CC [M]  core/rtw_chplan.o
  CC [M]  core/rtw_recv.o
  CC [M]  core/rtw_sta_mgt.o
  CC [M]  core/rtw_ap.o
  CC [M]  core/mesh/rtw_mesh.o
  CC [M]  core/mesh/rtw_mesh_pathtbl.o
  CC [M]  core/mesh/rtw_mesh_hwmp.o
  CC [M]  core/rtw_xmit.o
  CC [M]  core/rtw_p2p.o
  CC [M]  core/rtw_rson.o
  CC [M]  core/rtw_tdls.o
  CC [M]  core/rtw_br_ext.o
  CC [M]  core/rtw_iol.o
  CC [M]  core/rtw_sreset.o
  CC [M]  core/rtw_btcoex_wifionly.o
  CC [M]  core/rtw_btcoex.o
  CC [M]  core/rtw_beamforming.o
  CC [M]  core/rtw_odm.o
  CC [M]  core/rtw_rm.o
  CC [M]  core/rtw_rm_fsm.o
  CC [M]  core/rtw_rm_util.o
  CC [M]  core/efuse/rtw_efuse.o
  CC [M]  os_dep/osdep_service.o
  CC [M]  os_dep/linux/os_intfs.o
  CC [M]  os_dep/linux/usb_intf.o
  CC [M]  os_dep/linux/usb_ops_linux.o
  CC [M]  os_dep/linux/ioctl_linux.o
  CC [M]  os_dep/linux/xmit_linux.o
  CC [M]  os_dep/linux/mlme_linux.o
  CC [M]  os_dep/linux/recv_linux.o
  CC [M]  os_dep/linux/ioctl_cfg80211.o
  CC [M]  os_dep/linux/rtw_cfgvendor.o
  CC [M]  os_dep/linux/wifi_regd.o
os_dep/linux/ioctl_cfg80211.c:10216:25: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  int *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, int *)’ [-Wincompatible-pointer-types]
10216 |         .get_tx_power = cfg80211_rtw_get_txpower,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:10216:25: note: (near initialization for ‘rtw_cfg80211_ops.get_tx_power’)
make[4]: *** [/usr/lib/modules/6.14.1-arch1-1/build/scripts/Makefile.build:207: os_dep/linux/ioctl_cfg80211.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/lib/modules/6.14.1-arch1-1/build/Makefile:1994: .] Error 2
make[2]: *** [/usr/lib/modules/6.14.1-arch1-1/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/rtl88x2bu/5.8.7.1/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.14.1-arch1-1/build'
make: *** [Makefile:2395: modules] Error 2

# exit code: 2
# elapsed time: 00:00:20
----------------------------------------------------------------
`

MaxG87, 2025-06-08:
Kernel 6.14 requires a new patch. Please try with revision 00cd158900bf44c86edf589fd86852dc5003e08f or newer and report if the issue still exists.

closed 2025-06-08 by MaxG87

PaulStogov, 2025-06-12:
Works fine now even with 6.15.2. Thank you very much!