cilynx/rtl88x2bu #281: build on 6.16 fails

sirko-schroeder, 2025-08-01:
Hi,

building against a new 6.16 kernel throws:

././include/osdep_service_linux.h:357:26: error: implicit declaration of function ‘from_timer’; did you mean ‘mod_timer’? [-Wimplicit-function-declaration]

analog to https://bugs.launchpad.net/ubuntu/+source/broadcom-sta/+bug/2114261
(https://launchpadlibrarian.net/799929166/broadcom-sta_6.30.223.271-27ubuntu1.questing.debdiff)

I did the same for the from_timer in osdep_service_linux.h

diff --git a/include/osdep_service_linux.h b/include/osdep_service_linux.h
index 154e971..aacbeb7 100644
--- a/include/osdep_service_linux.h
+++ b/include/osdep_service_linux.h
@@ -353,7 +353,9 @@ static inline void timer_hdl(struct timer_list *in_timer)
 static inline void timer_hdl(unsigned long cntx)
 #endif
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 16, 0))
+        _timer *ptimer = timer_container_of(ptimer, in_timer, timer);
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 14, 0))
        _timer *ptimer = from_timer(ptimer, in_timer, timer);
 #else
        _timer *ptimer = (_timer *)cntx;

Now it builds again and is currently working here just fine.

Kind Regards,
Sirko

sirko-schroeder, 2025-08-01:
[from_timer_6.16.patch.txt](https://github.com/user-attachments/files/21541619/from_timer_6.16.patch.txt)

MaxG87, 2025-09-19:
I cannot reproduce that. Using CachyOS, I can build the driver using `make CC=clang LD=ld.lld`. I use Linux v6.16.8 (62dae0198231) and c3192d7153de4b324fd662bb9db3ce775879f5e5 of the driver.

closed 2025-09-19 by MaxG87